ngscopeclient/scopehal #2: Allow measurements to draw overlays

azonenberg, 2019-04-14:
Region being sampled by a measurement, etc

jwise, 2024-09-30:
Necro-ticket context: it would be nice to be able to draw wavedrom-style arrows on, for instance, the setup and hold measurements:

[image]

or it would be nice for a frequency measurement to be able to show where it is measuring, for instance, as shown below from my DHO4000:

[image]

or, analogously, measuring rise time:

[image]

jwise, 2024-09-30:
Presumably ought be moved to ngscopeclient/scopehal-apps, though.

azonenberg, 2024-09-30:
We should have a companion issue there, but the GUI side work is fairly minimal. The hard part is defining the scopehal-side APIs for filters to declare this sort of information that they wish to have displayed.