ngscopeclient/scopehal #11: Add Siglent SDS1204 driver

azonenberg, 2019-04-16:
Seems to be a Siglent OEM version of the LeCroy WaveAce. Almost the same SCPI command set as MAUI scopes, but via raw TCP instead of VICP.

azonenberg, 2019-05-02:
Per @four0four : there appears to be a firmware bug in which WF? queries occasionally are dropped and need to be re-sent.

four0four, 2019-05-04:
Another Siglent quirk is that sometimes it appears to crash if commands are sent too rapidly - mostly observed with INR. Not sure if this is just what happens to be sent the most or there's a queuing issue in Siglent's firmware. Either way, possible fixes are rate limiting INRs, or (better) slowing queries when retrying.

four0four, 2020-02-26:
Better understand the issues now: tl;dr is that the scopes will occasionally ACK a SCPI request and simply not fill it. We can work around this by adding a back-off timeout and retry to avoid the current situation, which results in deadlocks in the UI thread. 

Support pending workaround is available as of 99468ea, so I'm closing this.

closed 2020-02-26 by four0four