ngscopeclient/scopehal #12: Add Rigol scope driver(s)

azonenberg, 2019-04-16:


raplin, 2019-04-28:
Rigol scopes are lamentably slow for waveform downloads; extensive discussion here.
https://www.eevblog.com/forum/testgear/ethernet-transfer-speed-for-rigol-ds2072a-and-ds4014/
TL;DR; about 1.7MBytes/sec or less over ethernet or USB.
Somewhat restrictive; if scope is running you can only d/l the displayed waveform (e.g. 600pts); if stopped you can fetch the capture memory (24M samples (e.g. DS1054z full ram) takes 15+seconds.

azonenberg, 2019-04-28:
Most scopes are pretty slow over Ethernet/USB because they're not designed for fast downloads. Sadly, fixing this is likely to require custom hardware or a major change of priority at the scope vendors.

Typically glscopeclient will do a one-shot trigger, download the waveform, and repeat as needed.

azonenberg, 2019-05-04:
We can now download waveforms and do very basic read-only connections to a DS1104Z. No USB/RS232 connection support and all Set*() functions are still unimplemented.

closed 2020-05-26 by azonenberg