ngscopeclient/scopehal #15: Add Picoscope scope driver(s)

azonenberg, 2019-04-16:


bvernoux, 2019-04-25:
Does it is too early to start working on adding Picoscope ?
* After checking quickly the code it seems a good start will be to use class example like LeCroyOscilloscope.cpp/.h
* Does the Interface/Class with members... is frozen ? (maybe it is too early to work on Picosope driver ?)

noopwafel, 2020-06-07:
As discussed on IRC: I started hacking up a prototype SCPI bridge for my 3000-series Picoscope, with data going over a separate socket as in AntikernelLabsOscilloscope.

bvernoux, 2020-12-07:
@noopwafel Do you have any news about your prototype SCPI bridge for my 3000-series Picoscope ?
I have a PicoScope 3406DMSO so I'm very interested by this driver as it shall be very fast in theory over USB 3.0

noopwafel, 2020-12-08:
I got (very very) swamped by work, but getting back to this is now nearer the top of my todo (likely next week?). Feel free to ping me about it. I don't have (access to) an MSO, so would be nice to hack that in too.

bvernoux, 2020-12-12:
Yes everyone is very busy with tons of things to do and we do that on our spare time so there is no hurry
I'm very interested if you have something usable to test to help and even improve it for PicoScope 3000-series (especially I have the full version) and I could test also on Windows.

sbabic44, 2021-02-02:
I also have a PicoScope 3406DMSO and be willing to test...

azonenberg, 2021-02-02:
@sbabic44 Unfortunately Pico's driver model uses a separate SDK for each one of their families. In order to clean this up a bit and provide network transparency (a key feature for me since my main desk is about 20 meters from the lab bench), I'm using a bridge server (https://github.com/azonenberg/scopehal-pico-bridge) as the point of abstraction which currently only supports the 6000 series.

Once I've finished 6000 series support I'm definitely interested in supporting lower end models. Some refactoring will be needed though, as right now the ps6000d application contains a bunch of SCPI parsing code etc that should probably be shared across the different server binaries.

closed 2021-03-14 by azonenberg

azonenberg, 2021-03-14:
Closed this ticket as 6000 series support is at "minimum viable" level.

Feel free to file additional tickets for specific features. Requests for new instrument support should be filed against scopehal-pico-bridge instead, as the network protocol used by the PicoOscilloscope class is intended to be the same for all instruments.

commit b005d923 (2021-03-14): PicoOscilloscope: initial trigger support. Fixes #15.