ngscopeclient/scopehal #16: Add Saleae Logic driver

azonenberg, 2019-04-16:


raplin, 2019-04-28:
What reasoning is there for not using libsigrok? It seems like you'd get a ton of devices 'for free'

azonenberg, 2019-04-28:
This would contaminate the scopehal core with a GPL dependency which is absolutely unacceptable. Libscopehal must remain permissively licensed to allow/encourage commercial tools to integrate with it.

s09bQ5, 2019-05-31:
As long as you don't force that backend to be enabled during compilation, users who don't like the GPL can still build a BSD licensed version.

l29ah, 2021-06-28:
So @azonenberg would you accept a patch adding an opt-in libsigrok backend support to scopehal?

azonenberg, 2021-06-29:
I think the cleaner way to do the license boundary is by means of a socket bridge, similar to what we do with PicoScope stuff. Completely separate binary you can either install or not, and the libscopehal binary remains GPL-free in all cases.