ngscopeclient/scopehal #26: Add Hantek MSO5074 driver

azonenberg, 2019-04-16:


azonenberg, 2020-08-02:
Somebody requested this ages ago, but so far nobody has contributed a scope to develop on... Leaving the issue open for now.

joelpmichael, 2025-01-10:
Putting my hand up to do some development on this as discussed on fedi. I have a Hantek 6254BC USB scope to develop against, other Hantek scopes could be added as hardware becomes available using a similar development approach.

Hantek supplies a SDK which provides headers against a set of Windows DLLs. Initial analysis of these DLLs suggests that functions are a thin wrapper around direct IOCTL communication with the Windows device driver. Quick support could be implemented using vendor blobs only on Windows, and cross-platform support could be implemented by observing USB traffic in a controlled manner and implementing a driver based on the observed behaviour.

As discussed, I will take a similar approach to the PicoScope.