ngscopeclient/scopehal #115: Linking liblxi

tomverbeure, 2020-05-19:
I trying to create a SCPILxiTransport class, which relies on liblxi for the low level details.

I've updated the scopehal/CMakeLists.txt to add a dependency on liblxi as follows:

```
--- a/scopehal/CMakeLists.txt
+++ b/scopehal/CMakeLists.txt
@@ -1,6 +1,8 @@
 include_directories(${GTKMM_INCLUDE_DIRS} ${SIGCXX_INCLUDE_DIRS})
 link_directories(${GTKMM_LIBRARY_DIRS} ${SIGCXX_LIBRARY_DIRS})
 
+find_library(LXI_LIB lxi)
+
 set(SCOPEHAL_SOURCES
        base64.cpp
        scopehal.cpp
@@ -10,6 +12,7 @@ set(SCOPEHAL_SOURCES
        SCPITransport.cpp
        SCPISocketTransport.cpp
        VICPSocketTransport.cpp
+       SCPILxiTransport.cpp
        SCPIDevice.cpp
 
        Instrument.cpp
@@ -38,7 +41,7 @@ set(SCOPEHAL_SOURCES
 
 add_library(scopehal SHARED
        ${SCOPEHAL_SOURCES})
-target_link_libraries(scopehal ${SIGCXX_LIBRARIES} ${GTKMM_LIBRARIES} xptools log graphwidget yaml-cpp)
+target_link_libraries(scopehal ${SIGCXX_LIBRARIES} ${GTKMM_LIBRARIES} xptools log graphwidget yaml-cpp ${LXI_LIB})
```
libscopehal.so itself compiles fine, but when I try to build the applications, it fails because it doesn't link with liblxi:

```
[ 77%] Linking CXX executable glscopeclient
../../lib/scopehal/libscopehal.so: undefined reference to `lxi_init()'
collect2: error: ld returned 1 exit status
src/glscopeclient/CMakeFiles/glscopeclient.dir/build.make:426: recipe for target 'src/glscopeclient/glscopeclient' failed
```
This is despite the fact that `CMakeFiles/glscopeclient.dir/link.txt` has the following on the linking command:
```
/usr/lib/x86_64-linux-gnu/liblxi.so ../../lib/scopeprotocols/libscopeprotocols.so 
```

liblxi has been installed with a simply `apt install liblxi-dev`, and example programs outside of scopehal compile just fine.

I'm very new at cmake obviously. Any pointer on how to unblock this?

Tom

azonenberg, 2020-05-19:
Did you add liblxi to the target_link_libraries line in the glscopeclient CMakeLists or just libscopehal? You might need to link it with the application too.

tomverbeure, 2020-05-19:
I did. I now think the issue is not with cmake, it's with the linking command itself.

I took the generated link.txt file. It looks like this:

/usr/bin/c++  -g -fopenmp -Wall -Wextra -Wuninitialized  -Wshadow -Wunsafe-loop-optimizations -Wpedantic -Wcast-align -Wwrite-strings -Wmissing-declarations -Wvla --std=c++11 -mtune=native  -rdynamic CMakeFiles/glscopeclient.dir/ChannelPropertiesDialog.cpp.o CMakeFiles/glscopeclient.dir/Framebuffer.cpp.o CMakeFiles/glscopeclient.dir/HistoryWindow.cpp.o CMakeFiles/glscopeclient.dir/MeasurementDialog.cpp.o CMakeFiles/glscopeclient.dir/OscilloscopeWindow.cpp.o CMakeFiles/glscopeclient.dir/Program.cpp.o CMakeFiles/glscopeclient.dir/ProtocolAnalyzerWindow.cpp.o CMakeFiles/glscopeclient.dir/ProtocolDecoderDialog.cpp.o CMakeFiles/glscopeclient.dir/ScopeApp.cpp.o CMakeFiles/glscopeclient.dir/Shader.cpp.o CMakeFiles/glscopeclient.dir/ShaderStorageBuffer.cpp.o CMakeFiles/glscopeclient.dir/Texture.cpp.o CMakeFiles/glscopeclient.dir/Timeline.cpp.o CMakeFiles/glscopeclient.dir/TriggerPropertiesDialog.cpp.o CMakeFiles/glscopeclient.dir/VertexArray.cpp.o CMakeFiles/glscopeclient.dir/VertexBuffer.cpp.o CMakeFiles/glscopeclient.dir/WaveformArea.cpp.o CMakeFiles/glscopeclient.dir/WaveformArea_events.cpp.o CMakeFiles/glscopeclient.dir/WaveformArea_rendering.cpp.o CMakeFiles/glscopeclient.dir/WaveformArea_cairo.cpp.o CMakeFiles/glscopeclient.dir/WaveformGroup.cpp.o CMakeFiles/glscopeclient.dir/main.cpp.o  -o glscopeclient  -Wl,-rpath,/home/tom/projects/scopehal-cmake/build/lib/scopehal:/home/tom/projects/scopehal-cmake/build/lib/scopemeasurements:/home/tom/projects/scopehal-cmake/build/lib/graphwidget:/home/tom/projects/scopehal-cmake/build/lib/scopeprotocols **/usr/lib/x86_64-linux-gnu/liblxi.so** ../../lib/scopehal/libscopehal.so ../../lib/scopemeasurements/libscopemeasurements.so -lgtkmm-3.0 -latkmm-1.6 -lgdkmm-3.0 -lgiomm-2.4 -lpangomm-1.4 -lglibmm-2.4 -lgtk-3 -lgdk-3 -lpangocairo-1.0 -lpango-1.0 -latk-1.0 -lcairo-gobject -lgio-2.0 -lcairomm-1.0 -lcairo -lsigc-2.0 -lgdk_pixbuf-2.0 -lgobject-2.0 -lglib-2.0 -lsigc-2.0 -lGL -lyaml-cpp ../../lib/xptools/libxptools.a ../../lib/log/liblog.a ../../lib/graphwidget/libgraphwidget.so -lgtkmm-3.0 -latkmm-1.6 -lgdkmm-3.0 -lgiomm-2.4 -lpangomm-1.4 -lglibmm-2.4 -lgtk-3 -lgdk-3 -lpangocairo-1.0 -lpango-1.0 -latk-1.0 -lcairo-gobject -lgio-2.0 -lcairomm-1.0 -lcairo -lsigc-2.0 -lgdk_pixbuf-2.0 -lgobject-2.0 -lglib-2.0 ../../lib/scopeprotocols/libscopeprotocols.so /usr/local/lib/libffts.a

So liblxi is listed as a library, yet it still fails:
```
../../lib/scopehal/libscopehal.so: undefined reference to `lxi_init()'
collect2: error: ld returned 1 exit status
```
I get the same if I replace the bold stuff above with `-llxi`.

So there is some issue with linking that eludes me.

Tom

miek, 2020-05-19:
I think the problem is the ordering in that command, liblxi.so should come after libscopehal.so (see here for why: https://eli.thegreenplace.net/2013/07/09/library-order-in-static-linking). I don't know how to get CMake to do that though.

tomverbeure, 2020-05-21:
So I don't think it's a CMake issue.

When you run `ldd libscopehal.so`, you get a very long list with all the dependent libraries. This list is essentially all the libraries that were given in link.txt 2 comments earlier.

For one reason or the other, liblxi is not added to the list. 

Next is to try to find out why what is... I've already tried recompiling liblxi from source instead of using the liblxi that comes with Ubuntu, but that didn't solve it either.

Trying more things...

Tom

tomverbeure, 2020-05-24:
Things magically started working after rebooting PC. It think it was a matter of reloading the shared library cache?

Closing...

closed 2020-05-24 by tomverbeure