ngscopeclient/scopehal #175: Crash with Rigol DS1104Z connected via USBTMC

whitequark, 2020-07-09:
Scope string: `Rigol DS1104Z:rigol:usbtmc:/dev/usbtmc2` (no idea why it's usbtmc2)

Backtrace:
```
Thread 7 "ScopeThread" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffe9d55700 (LWP 249288)]
0x00007ffff657fb0f in RigolOscilloscope::AcquireData (this=0x555555ae1c00, toQueue=true)
    at /home/whitequark/Projects/scopehal-cmake/lib/scopehal/RigolOscilloscope.cpp:631
631                                     s[m_channels[j]] = pending_waveforms[j][i];
(gdb) bt
#0  0x00007ffff657fb0f in RigolOscilloscope::AcquireData(bool) (this=0x555555ae1c00, toQueue=true)
    at /home/whitequark/Projects/scopehal-cmake/lib/scopehal/RigolOscilloscope.cpp:631
azonenberg/scopehal-apps#1  0x00005555556e0c3c in ScopeThread(Oscilloscope*) (scope=0x555555ae1c00)
    at /home/whitequark/Projects/scopehal-cmake/src/glscopeclient/main.cpp:358
azonenberg/scopehal-apps#2  0x00005555556a26fd in std::__invoke_impl<void, void (*)(Oscilloscope*), Oscilloscope*>(std::__invoke_other, void (*&&)(Oscilloscope*), Oscilloscope*&&) (__f=@0x5555559c7b40: 0x5555556e0a3b <ScopeThread(Oscilloscope*)>, __args#0=@0x5555559c7b38: 0x555555ae1c00)
    at /usr/include/c++/8/bits/invoke.h:60
azonenberg/scopehal-apps#3  0x00005555556a2218 in std::__invoke<void (*)(Oscilloscope*), Oscilloscope*>(void (*&&)(Oscilloscope*), Oscilloscope*&&)
    (__fn=@0x5555559c7b40: 0x5555556e0a3b <ScopeThread(Oscilloscope*)>, __args#0=@0x5555559c7b38: 0x555555ae1c00)
    at /usr/include/c++/8/bits/invoke.h:95
azonenberg/scopehal-apps#4  0x00005555556a35cd in std::thread::_Invoker<std::tuple<void (*)(Oscilloscope*), Oscilloscope*> >::_M_invoke<0ul, 1ul>(std::_Index_tuple<0ul, 1ul>) (this=0x5555559c7b38) at /usr/include/c++/8/thread:244
azonenberg/scopehal-apps#5  0x00005555556a3588 in std::thread::_Invoker<std::tuple<void (*)(Oscilloscope*), Oscilloscope*> >::operator()() (this=0x5555559c7b38)
    at /usr/include/c++/8/thread:253
azonenberg/scopehal-apps#6  0x00005555556a356c in std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (*)(Oscilloscope*), Oscilloscope*> > >::_M_run()
    (this=0x5555559c7b30) at /usr/include/c++/8/thread:196
azonenberg/scopehal-apps#7  0x00007ffff633fbe0 in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
azonenberg/scopehal-apps#8  0x00007ffff60b7f27 in start_thread (arg=<optimized out>) at pthread_create.c:479
azonenberg/scopehal-apps#9  0x00007ffff5fe92ef in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

azonenberg, 2020-07-09:
Moved to scopehal since this is a driver bug, not a UI bug

noopwafel, 2020-07-30:
this happens when an incomplete set of waveforms ends up being pushed (e.g. when your maxpoints is 0 for some reason); can't work out how this would happen in the current codebase, but if it's still broken, #202 might fix it

azonenberg, 2020-10-25:
Closing because no activity in several months and believed to be fixed. Reopen if this still happens.

closed 2020-10-25 by azonenberg