ngscopeclient/scopehal #222: Streaming devices support/integration

smunaut, 2020-08-07:
Some capture devices don't have an acquisition buffer so it's not the classic "trigger / capture / download" cycle, but it's just start acquisition then you get samples until you said you had enough.

With sample compression this can also be quite high rate / over long periods (so raw uncompressed samples could be very memory hungry). Like for instance if you monitor a serial bus that's idle most of the time, and you have high oversampling just to get more time resolution.

sslupsky, 2024-04-13:
There is a project that does this that is quite interesting:
https://github.com/pico-coder/sigrok-pico

The device is a Raspberry Pi Pico with some custom PIO firmware that enables logic analyzer capture up to 120 Msps and analog capture.  The device implements a streaming protocol with RLL compression to maximize bandwidth across a USB 2.0 Full Speed interface.

ohault, 2025-03-17:
I would find interesting for beginners and makers to have a support for Raspberry Pi Pico 2 and Pico 2 W.

ohault, 2025-03-20:
> You can utilize this by set the buffer length in the gui, and grap the stream and buffer it on pc side...
> Just start without trigger. or even add a trigger signal to start....
> I did it in a way there scopehal starts a binary stream when it wants data and stops it when the selected samples has recieved...
> be aware on the limitations of speed on the usb... for logic analyser there is not soo much data to transfer, but for analog data the sample rate can not be too high... but decent for something.

Hi @rapzak, I meant native support for Pico 2 and Pico 2 W, with a directly usable firmware image with ngscopeclient.