ngscopeclient/scopehal #230: Wrong FFTS include path

smunaut, 2020-08-24:
`DeEmbedDecoder.h` and `FFTDecoder.h` use `#include <ffts/ffts.h>` instead of `#include <ffts.h>`.

Some other places do it correctly and if you look at how ` FindFFTS.cmake` works, `LIBFFTS_INCLUDE_DIR`  will already include the `ffts/` component.

But since `glscopeclient` includes something that includes those, you also need to simultaneously add `LIBFFTS_INCLUDE_DIR` to the `include_directories` of that application.

azonenberg, 2020-08-24:
We've had issues where on Windows, the opposite is true: ffts.h fails to compile and ffts/ffts.h works.

Maybe FindFFTS.cmake needs porting work to find the library on Windows?

azonenberg, 2020-08-24:
(paging @nshcat to look into this)

smunaut, 2020-08-24:
ATM some source file use `#include <ffts.h>`

```
scopeprotocols/CTLEDecoder.cpp:#include <ffts.h>
scopeprotocols/ChannelEmulationDecoder.cpp:#include <ffts.h>
```

So if those build the issue with windows should be elsewhere.

As mentionned I had to update the apps CMakelist.txt because the includes in headers will will leak there.

```diff
--- a/src/glscopeclient/CMakeLists.txt
+++ b/src/glscopeclient/CMakeLists.txt
@@ -11,7 +11,7 @@ if(WIN32)
 endif()
 
 #Set up include paths
-include_directories(${GTKMM_INCLUDE_DIRS} ${SIGCXX_INCLUDE_DIRS} ${GLEW_INCLUDE_DIRS})
+include_directories(${GTKMM_INCLUDE_DIRS} ${SIGCXX_INCLUDE_DIRS} ${GLEW_INCLUDE_DIRS} ${LIBFFTS_INCLUDE_DIR})
 link_directories(${GTKMM_LIBRARY_DIRS} ${SIGCXX_LIBRARY_DIRS})
 
 ###############################################################################
```

smunaut, 2020-08-24:
Actually a better solution is to set the FFTS include dir in the `PUBLIC` include of the `scopeprotocols` since it's required by anything using it.

```
diff --git a/scopeprotocols/CMakeLists.txt b/scopeprotocols/CMakeLists.txt
index e604b34..4c637f3 100644
--- a/scopeprotocols/CMakeLists.txt
+++ b/scopeprotocols/CMakeLists.txt
@@ -86,6 +86,6 @@ target_link_libraries(scopeprotocols
 
 target_include_directories(scopeprotocols
 PUBLIC ${CMAKE_CURRENT_SOURCE_DIR}
-PRIVATE ${LIBFFTS_INCLUDE_DIR})
+       ${LIBFFTS_INCLUDE_DIR})
 
 install(TARGETS scopeprotocols LIBRARY DESTINATION /usr/lib)
```

azonenberg, 2020-08-24:
I'm in the middle of a major refactoring right now.

Send a PR in a day or so when I'm done? I don't want you to have to put too much effort into merging/rebasing.

azonenberg, 2020-08-26:
Done with the refactoring if you want to look at this. Some of the FFTS includes may have got twiddled as part of this, so probably best to go check everywhere.

PR #231 (2020-08-30): Fix ffts include path
From how FindFFTS.cmake works, LIBFFTS_INCLUDE_DIR will already
include the ffts/ component.

Fixes #230
commit 863ee1dc (2020-08-30): Fix ffts include path

From how FindFFTS.cmake works, LIBFFTS_INCLUDE_DIR will already
include the ffts/ component.

Fixes #230
closed 2020-08-30 by azonenberg