ngscopeclient/scopehal #260: Issue to build the actual code with MSYS2 / MINGW64 related to LeCroyOscilloscope.cpp

bvernoux, 2020-09-21:
Issue to build the actual code with MSYS2 / MINGW64 related to TouchstoneParser.cpp code
-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
...
Error
```
[ 26%] Building CXX object lib/scopehal/CMakeFiles/scopehal.dir/LeCroyOscilloscope.cpp.obj
C:\msys64\home\Ben\scopehal-apps\lib\scopehal\LeCroyOscilloscope.cpp: In member function 'bool LeCroyOscilloscope::ReadWavedescs(std::vector<std::__cxx11::basic_string<char> >&, bool*, unsigned int&, bool&)':
C:\msys64\home\Ben\scopehal-apps\lib\scopehal\LeCroyOscilloscope.cpp:1646:31: warning: unknown conversion type character 'z' in format [-Wformat=]
 1646 |    LogError("Got wavedesc of %zu bytes (expected %zu)\n", wavedescs[i].size(), expected_wavedesc_size);
      |                               ^
C:\msys64\home\Ben\scopehal-apps\lib\scopehal\LeCroyOscilloscope.cpp:1646:51: warning: unknown conversion type character 'z' in format [-Wformat=]
 1646 |    LogError("Got wavedesc of %zu bytes (expected %zu)\n", wavedescs[i].size(), expected_wavedesc_size);
      |                                                   ^
C:\msys64\home\Ben\scopehal-apps\lib\scopehal\LeCroyOscilloscope.cpp:1646:13: warning: too many arguments for format [-Wformat-extra-args]
 1646 |    LogError("Got wavedesc of %zu bytes (expected %zu)\n", wavedescs[i].size(), expected_wavedesc_size);
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\msys64\home\Ben\scopehal-apps\lib\scopehal\LeCroyOscilloscope.cpp: In member function 'std::map<int, Waveform<EmptyConstructorWrapper<bool> >*> LeCroyOscilloscope::ProcessDigitalWaveform(std::string&)':
C:\msys64\home\Ben\scopehal-apps\lib\scopehal\LeCroyOscilloscope.cpp:2015:33: warning: format '%ld' expects argument of type 'long int*', but argument 3 has type 'int64_t*' {aka 'long long int*'} [-Wformat=]
 2015 |  if(1 != sscanf(tmp.c_str(), "%ld", &timestamp))
      |                               ~~^   ~~~~~~~~~~
      |                                 |   |
      |                                 |   int64_t* {aka long long int*}
      |                                 long int*
      |                               %lld
C:\msys64\home\Ben\scopehal-apps\lib\scopehal\LeCroyOscilloscope.cpp:2023:2: error: 'localtime_r' was not declared in this scope; did you mean 'localtime_s'?
 2023 |  localtime_r(&tnow, &now);
      |  ^~~~~~~~~~~
      |  localtime_s
C:\msys64\home\Ben\scopehal-apps\lib\scopehal\LeCroyOscilloscope.cpp: In member function 'virtual uint64_t LeCroyOscilloscope::GetSampleRate()':
C:\msys64\home\Ben\scopehal-apps\lib\scopehal\LeCroyOscilloscope.cpp:2806:28: warning: format '%ld' expects argument of type 'long int*', but argument 3 has type 'int64_t*' {aka 'long long int*'} [-Wformat=]
 2806 |   sscanf(reply.c_str(), "%ld", &m_sampleRate);
      |                          ~~^   ~~~~~~~~~~~~~
      |                            |   |
      |                            |   int64_t* {aka long long int*}
      |                            long int*
      |                          %lld
C:\msys64\home\Ben\scopehal-apps\lib\scopehal\LeCroyOscilloscope.cpp: In member function 'virtual void LeCroyOscilloscope::SetSampleDepth(uint64_t)':
C:\msys64\home\Ben\scopehal-apps\lib\scopehal\LeCroyOscilloscope.cpp:2834:37: warning: format '%ld' expects argument of type 'long int', but argument 4 has type 'uint64_t' {aka 'long long unsigned int'} [-Wformat=]
 2834 |  snprintf(tmp, sizeof(tmp), "MSIZ %ld", depth);
      |                                   ~~^   ~~~~~
      |                                     |   |
      |                                     |   uint64_t {aka long long unsigned int}
      |                                     long int
      |                                   %lld
mingw32-make[2]: *** [lib\scopehal\CMakeFiles\scopehal.dir\build.make:545: lib/scopehal/CMakeFiles/scopehal.dir/LeCroyOscilloscope.cpp.obj] Error 1
mingw32-make[1]: *** [CMakeFiles\Makefile2:251: lib/scopehal/CMakeFiles/scopehal.dir/all] Error 2
mingw32-make: *** [Makefile:149: all] Error 2
```

azonenberg, 2020-09-22:
@bvernoux Try adding -D_POSIX_THREAD_SAFE_FUNCTIONS to the Windows compile flags in scopehal-apps/CMakeLists.txt:15 and see if that helps.

bvernoux, 2020-09-22:
I confirm adding -D_POSIX_THREAD_SAFE_FUNCTIONS to the Windows compile flags in scopehal-apps/CMakeLists.txt:15 fix that issue

azonenberg, 2020-09-24:
Is this issue fixed in the current HEAD? If so, please close it.

bvernoux, 2020-09-24:
Waiting merge of Fix issue PR#260 azonenberg/scopehal-apps#185

bvernoux, 2020-09-24:
It is fixed now

closed 2020-09-24 by bvernoux