ngscopeclient/scopehal #262: Issue to build the actual code with MSYS2 / MINGW64 related to SignalGeneratorOscilloscope.cpp

bvernoux, 2020-09-22:
Issue to build the actual code with MSYS2 / MINGW64 related to SignalGeneratorOscilloscope.cpp code
-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
...
Error
```
[ 32%] Building CXX object lib/scopehal/CMakeFiles/scopehal.dir/RohdeSchwarzHMC8012Multimeter.cpp.obj
C:\msys64\home\Ben\scopehal-apps\lib\scopehal\SignalGeneratorOscilloscope.cpp: In member function 'virtual bool SignalGeneratorOscilloscope::AcquireData()':
C:\msys64\home\Ben\scopehal-apps\lib\scopehal\SignalGeneratorOscilloscope.cpp:120:2: error: 'usleep' was not declared in this scope; did you mean 'Glib::usleep'?
  120 |  usleep(40 * 1000);
      |  ^~~~~~
      |  Glib::usleep
In file included from C:/msys64/mingw64/include/glibmm-2.4/glibmm.h:143,
                 from C:/msys64/mingw64/include/gtkmm-3.0/gtkmm.h:87,
                 from C:/msys64/home/Ben/scopehal-apps/lib/graphwidget/Graph.h:39,
                 from C:\msys64\home\Ben\scopehal-apps\lib\scopehal\scopehal.h:50,
                 from C:\msys64\home\Ben\scopehal-apps\lib\scopehal\SignalGeneratorOscilloscope.cpp:36:
C:/msys64/mingw64/include/glibmm-2.4/glibmm/timer.h:72:6: note: 'Glib::usleep' declared here
   72 | void usleep(unsigned long microseconds);
      |      ^~~~~~
mingw32-make[2]: *** [lib\scopehal\CMakeFiles\scopehal.dir\build.make:601: lib/scopehal/CMakeFiles/scopehal.dir/SignalGeneratorOscilloscope.cpp.obj] Error 1
mingw32-make[2]: *** Waiting for unfinished jobs....
mingw32-make[1]: *** [CMakeFiles\Makefile2:251: lib/scopehal/CMakeFiles/scopehal.dir/all] Error 2
mingw32-make: *** [Makefile:149: all] Error 2
```
Issue is related to usleep() which is deprecated since POSIX.1-2001 Issue 6 and not supported by MinGW
See https://sourceforge.net/p/mingw/mailman/message/33179314/
Potential solution for that https://gist.github.com/ngryman/6482577

bvernoux, 2020-09-22:
usleep() is used in following files:
1) scopehal-apps\lib\scopehal\Oscilloscope.cpp (1 hit)
	Line 152: 		usleep(10 * 1000);
2) scopehal-apps\lib\scopehal\SignalGeneratorOscilloscope.cpp (1 hit)
	Line 120: 	usleep(40 * 1000);
3) scopehal-apps\src\examples\curvetrace\main.cpp (2 hits)
	Line 107: 		usleep(25 * 1000);
	Line 121: 	usleep(50 * 1000);
4) scopehal-apps\src\examples\fgtest\main.cpp (1 hit)
	Line 87: 			usleep(1000 * 50);
5) scopehal-apps\src\glscopeclient\OscilloscopeWindow.cpp (1 hit)
	Line 792: 			usleep(1000 * 50);

nshcat, 2020-09-22:
It is better to use `std::this_thread::sleep(std::chrono::microseconds(..))` than the WinAPI stuff you mentioned, since it is cross-platform.

bvernoux, 2020-09-23:
I have fixed it partially as it needs to be modified in other repository but waiting merge of https://github.com/azonenberg/scopehal-apps/pull/183 too

closed 2020-09-24 by bvernoux