ngscopeclient/scopehal #296: Demo oscilloscope

bluecmd, 2020-10-05:
Hi,

I was checking out glscopeclient but I do not have any of the supported scopes. I was mostly curious about how it looks and feels to play with, so I was hoping that it had some sort of demo-mode where you get some generated stream of input and you can play around with the tools.

I found MockOscilloscope and it is close to what I'm suggesting, but does not appear to ever return any data, nor can it be selected in the glscopeclient driver configuration pop-up.

azonenberg, 2020-10-05:
There's currently no demo scope. SignalGeneratorOscilloscope is close, but is currently hard coded to a 1.25 Gbps PRBS-31 through one specific IBIS model from a specific file on my workstation. We haven't figured out the details of the UI for configuring it yet.

MockOscilloscope is not intended as a demo, it's used for offline display of saved waveform files. Since glscopeclient always expects to be connected to an instrument, we needed a shim that never triggers or does anything, but implements the Oscilloscope API, in order to keep it happy.

Would you be happy with just a noisy sinewave on a couple analog channels, or would you want a more complex data stream including some digital data etc?

bluecmd, 2020-10-05:
> Would you be happy with just a noisy sinewave on a couple analog channels, or would you want a more complex data stream including some digital data etc?

That would certainly be a good start to just mess around with the program.

> or would you want a more complex data stream including some digital data etc?

That would of course be cool to let me demo the decoders, but I can understand that being a v2.0 feature :-).

commit c9ed93b4 (2020-10-06): Added demo oscilloscope. Fixes #296.
closed 2020-10-06 by azonenberg