ngscopeclient/scopehal #331: Channel names not populating in "Trigger properties" dialogue (Rigol DS1054Z)

Cushychicken, 2020-10-31:
The "din" menu is popping up empty for me. Running Ubuntu 20.04, connected over USBTMC to a Rigol DS1054Z.

Looks like the "din" channel name list has the right number of entries (one blank for no channel selected, and four more - one for each analog scope channel), but no names associated with them. 

[image]

Perhaps `GetDisplayName()` is turning up blank for Rigol scopes? (Happy to check this behind the scenes if there's means of accessing that via CLI) 

https://github.com/azonenberg/scopehal-apps/blob/3752d10fc8a21ea60da8b81ed4c2f7ed17b7e665/src/glscopeclient/TriggerPropertiesDialog.cpp#L208

azonenberg, 2020-10-31:
Can you provide a screenshot of the main glscopeclient window without trigger properties open?

Cushychicken, 2020-10-31:
Howzat? 

[image]

azonenberg, 2020-10-31:
There's supposed to be channel label buttons in the bottom left of each channel, as in this screenshot.

What's interesting is that I'm not seeing sample rate information either. If you double-click that tiny blob do you get a channel properties dialog? And if you enter a name for the channel does it act somewhat more sane?

![100basetx-eyemask](https://user-images.githubusercontent.com/998036/97779445-5780a080-1b3b-11eb-978f-
8ef0e6ded34d.png)

Cushychicken, 2020-10-31:
Yep - I just see blank entries for all of those by default. 

I _am_ able to click those label buttons and edit the channel name: 

[image]

And that channel name, in turn, appears to propagate to the trigger properties:

[image]

(Also please tell me you're in Europe or somewhere - I'd hate to think I woke you up at quarter to 6 PST to talk trigger properties)

azonenberg, 2020-10-31:
I'm in the Seattle area, but my sleep schedule is destroyed from not having to wake up at a consistent time to go into the office for the past 6+ months :P I've actually been up since around 4AM Pacific time.

So what this tells me is that the glscopeclient UI is correctly handling an empty channel name, and the Rigol driver is incorrectly supplying an empty name by default. When you open the channel properties dialog, under "scope" and "channel" do you see plausible values, or is the "channel" line above "display name" blank?

Cushychicken, 2020-10-31:
`scope` and `channel` both appear like plausible values:

[image]

I'm guessing you'd expect something like `CHAN1` to be the default channel name, and also show up in the Trigger properties dialogue? 

> I'm in the Seattle area, but my sleep schedule is destroyed from not having to wake up at a consistent time to go into the office for the past 6+ months :P I've actually been up since around 4AM Pacific time.

That eases my mind a bit - I don't want to feel like I'm dragging you out of bed at godawful o'clock for this! You're certainly not alone on the sleeplessness front. I can't remember the last time I slept past 6 AM. How on earth is not having a regular job schedule making me wake up _earlier?!_

azonenberg, 2020-10-31:
Very interesting, so the hardware channel name is correct. The display name *should* default to the hardware channel name. I'm gonna have to do some digging here.

Cushychicken, 2020-10-31:
Anything I can do to help? 

As an aside - I read thru the manual and saw some references to Siglent's scopes being based off of Teledyne Lecroy firmware. Would you expect Siglent's scopes to have better functionality with glscopeclient?

(I'd love to pick your brain about this via email - got some questions I don't want to spam GitHub comments with - email me? You can reach me at <my GitHub handle> @ gmail.com)

azonenberg, 2020-10-31:
IRC is best, join #scopehal on Freenode? I have to run and take care of somethign but will be back at my desk in half an hour or so.

Cushychicken, 2020-10-31:
Never tried Freenode but there's a first time for everything - see you on there :+1:

azonenberg, 2020-10-31:
1af3ff87851660775578abf347a1680ecbda9105 should fix it, please test and close this issue if you can confirm the fix.

Cushychicken, 2020-10-31:
Fix confirmed! See screenshot below. Closing.  

[image]

closed 2020-10-31 by Cushychicken