ngscopeclient/scopehal #341: sds1104x-e doesn't work

RX14, 2020-11-15:
I attempted to connect to an sds1104x-e with glscope client and got the following trace:

```
$ build/src/glscopeclient/glscopeclient --debug --trace SCPISocketTransport siglent:siglent:lan:192.168.0.232:5025
Warning: glscopeclient works best with the OMP_WAIT_POLICY environment variable set to PASSIVE
Detecting CPU features...
    * AVX2
    
Connecting to SCPI oscilloscope at 192.168.0.232:5025
[SCPISocketTransport::SendCommand] Sending *IDN?
[SCPISocketTransport::ReadReply] Got Siglent Technologies,SDS1104X-E,SDSMMEBC4R1275,8.1.6.1.35R2
[SCPISocketTransport::SendCommand] Sending CHDR OFF
[SCPISocketTransport::SendCommand] Sending *IDN?
[SCPISocketTransport::ReadReply] Got Siglent Technologies,SDS1104X-E,SDSMMEBC4R1275,8.1.6.1.35R2
Warning: Model "SDS1104X-E" is unknown, available sample rates/memory depths may not be properly detected
[SCPISocketTransport::SendCommand] Sending COMM_FORMAT DEF9,BYTE,BIN
[SCPISocketTransport::SendCommand] Sending VBS 'app.Acquisition.Horizontal.Maximize="FixedSampleRate"'
[SCPISocketTransport::SendCommand] Sending COMBINE_CHANNELS?
Warning: Socket read failed (errno=11)
[SCPISocketTransport::ReadReply] Got 
[SCPISocketTransport::SendCommand] Sending COMBINE_CHANNELS 1
[SCPISocketTransport::SendCommand] Sending INR?
[SCPISocketTransport::ReadReply] Got 8193

[SCPISocketTransport::SendCommand] Sending *OPT?
Warning: Socket read failed (errno=11)
[SCPISocketTransport::ReadReply] Got 

[SCPISocketTransport::SendCommand] Sending VBS? 'return = Len(app.Acquisition.C1.Alias)'
Warning: Socket read failed (errno=11)
[SCPISocketTransport::ReadReply] Got 
[SCPISocketTransport::SendCommand] Sending VBS? 'return = app.Acquisition.C1.Alias'
Warning: Socket read failed (errno=11)
[SCPISocketTransport::ReadReply] Got 
[SCPISocketTransport::SendCommand] Sending VBS? 'return = Len(app.Acquisition.C2.Alias)'
Warning: Socket read failed (errno=11)
[SCPISocketTransport::ReadReply] Got 
[SCPISocketTransport::SendCommand] Sending VBS? 'return = app.Acquisition.C2.Alias'
Warning: Socket read failed (errno=11)
[SCPISocketTransport::ReadReply] Got 
[SCPISocketTransport::SendCommand] Sending VBS? 'return = Len(app.Acquisition.C3.Alias)'
Warning: Socket read failed (errno=11)
[SCPISocketTransport::ReadReply] Got 
[SCPISocketTransport::SendCommand] Sending VBS? 'return = app.Acquisition.C3.Alias'
Warning: Socket read failed (errno=11)
[SCPISocketTransport::ReadReply] Got 
[SCPISocketTransport::SendCommand] Sending VBS? 'return = Len(app.Acquisition.C4.Alias)'
Warning: Socket read failed (errno=11)
[SCPISocketTransport::ReadReply] Got 
[SCPISocketTransport::SendCommand] Sending VBS? 'return = app.Acquisition.C4.Alias'
Warning: Socket read failed (errno=11)
[SCPISocketTransport::ReadReply] Got 
[SCPISocketTransport::SendCommand] Sending C1:TRACE ON
[SCPISocketTransport::SendCommand] Sending C2:TRACE ON
[SCPISocketTransport::SendCommand] Sending C3:TRACE ON
[SCPISocketTransport::SendCommand] Sending C4:TRACE ON
```

The programming guide for sds1/2xxx series scopes is available at https://siglentna.com/wp-content/uploads/2020/04/ProgrammingGuide_PG01-E02C.pdf

azonenberg, 2020-11-26:
It appears that the LeCroy and Siglent drivers have diverged more than I thought.

We probably need to write a new Siglent driver that doesn't reuse any of the LeCroy code in order to handle the non-Windows CE based Siglent scopes.

azonenberg, 2021-04-18:
Should be fixed in 8a88a068d61a1dd3028314f0674b74e83007bf8d.

closed 2021-04-18 by azonenberg