ngscopeclient/scopehal #358: glscopeclient hangs on Rigol MSO5000

pepijndevos, 2020-12-03:
When I configure the glscopeclient for my Rigol and press OK, it pops up a window if I want to kill the app or wait for it to respond, and then after a good few seconds the UI comes up.

It is extremely slow to load any data, showing like 3 WFMs, 0.05 WFM/s, and while the label of the sample rate matches what i see on the scope (2GSa/s, 4Mpts), the view is zoomed in to to 20.000 ns, while the scope shows 200us/div. One time I managed to zoom out, but generally the UI just hangs and the time scale vanishes completely when zooming too far.

I tried to trace the commands it sends:

```
$ ./src/glscopeclient/glscopeclient --debug --trace SCPISocketTransport
Connecting to SCPI oscilloscope at 169.254.67.109:5555
[SCPISocketTransport::SendCommand] Sending *IDN?
[SCPISocketTransport::ReadReply] Got RIGOL TECHNOLOGIES,MSO5074,MS5A221301338,00.01.03.00.01
[SCPISocketTransport::SendCommand] Sending SYST:OPT:STAT? RL2
[SCPISocketTransport::ReadReply] Got 0
[SCPISocketTransport::SendCommand] Sending WAV:FORM BYTE
[SCPISocketTransport::SendCommand] Sending WAV:MODE RAW
[SCPISocketTransport::SendCommand] Sending CHAN1:VERN ON
[SCPISocketTransport::SendCommand] Sending CHAN2:VERN ON
[SCPISocketTransport::SendCommand] Sending CHAN3:VERN ON
[SCPISocketTransport::SendCommand] Sending CHAN4:VERN ON
[SCPISocketTransport::SendCommand] Sending TIM:VERN ON
[SCPISocketTransport::SendCommand] Sending CHAN1:DISP ON
[SCPISocketTransport::SendCommand] Sending CHAN2:DISP ON
[SCPISocketTransport::SendCommand] Sending CHAN3:DISP ON
[SCPISocketTransport::SendCommand] Sending CHAN4:DISP ON
[SCPISocketTransport::SendCommand] Sending SING
[SCPISocketTransport::SendCommand] Sending TRIG:STAT?
[SCPISocketTransport::ReadReply] Got TD
[SCPISocketTransport::SendCommand]     Sending ACQ:MDEP?
[SCPISocketTransport::ReadReply]     Got 4.0000E+06
[SCPISocketTransport::SendCommand]     Sending WAV:SOUR CHAN1
[SCPISocketTransport::SendCommand]     Sending WAV:PRE?
[SCPISocketTransport::ReadReply]     Got 0,2,4000000,1,5.000000E-10,-1.000000E-3,0.000000,3.9962E-02,0,128
[SCPISocketTransport::SendCommand]     Sending WAV:STAR 1
[SCPISocketTransport::SendCommand]     Sending WAV:STOP 4000000
[SCPISocketTransport::SendCommand]     Sending WAV:DATA?
[SCPISocketTransport::SendCommand]     Sending WAV:SOUR CHAN2
[SCPISocketTransport::SendCommand]     Sending WAV:PRE?
[SCPISocketTransport::ReadReply]     Got 0,2,4000000,1,5.000000E-10,-1.000000E-3,0.000000,3.9832E-02,0,128
[SCPISocketTransport::SendCommand]     Sending WAV:STAR 1
[SCPISocketTransport::SendCommand]     Sending WAV:STOP 4000000
[SCPISocketTransport::SendCommand]     Sending WAV:DATA?
[SCPISocketTransport::SendCommand]     Sending WAV:SOUR CHAN3
[SCPISocketTransport::SendCommand]     Sending WAV:PRE?
[SCPISocketTransport::ReadReply]     Got 0,2,4000000,1,5.000000E-10,-1.000000E-3,0.000000,3.9620E-02,0,128
[SCPISocketTransport::SendCommand]     Sending WAV:STAR 1
[SCPISocketTransport::SendCommand]     Sending WAV:STOP 4000000
[SCPISocketTransport::SendCommand]     Sending WAV:DATA?
[SCPISocketTransport::SendCommand]     Sending WAV:SOUR CHAN4
[SCPISocketTransport::SendCommand]     Sending WAV:PRE?
[SCPISocketTransport::ReadReply]     Got 0,2,4000000,1,5.000000E-10,-1.000000E-3,0.000000,3.9927E-02,0,128
[SCPISocketTransport::SendCommand]     Sending WAV:STAR 1
[SCPISocketTransport::SendCommand]     Sending WAV:STOP 4000000
[SCPISocketTransport::SendCommand]     Sending WAV:DATA?
[SCPISocketTransport::SendCommand]     Sending SING
[SCPISocketTransport::SendCommand]     Sending CHAN1:SCALE?
[SCPISocketTransport::ReadReply]     Got 1
[SCPISocketTransport::SendCommand]     Sending CHAN1:OFFS?
[SCPISocketTransport::ReadReply]     Got 0
[SCPISocketTransport::SendCommand]     Sending TRIG:EDGE:SOUR?
[SCPISocketTransport::ReadReply]     Got CHAN4
    Trigger source: CHAN4
[SCPISocketTransport::SendCommand]     Sending CHAN2:SCALE?
[SCPISocketTransport::ReadReply]     Got 1
[SCPISocketTransport::SendCommand]     Sending CHAN2:OFFS?
[SCPISocketTransport::ReadReply]     Got 0
[SCPISocketTransport::SendCommand]     Sending CHAN3:SCALE?
[SCPISocketTransport::ReadReply]     Got 1
[SCPISocketTransport::SendCommand]     Sending CHAN3:OFFS?
[SCPISocketTransport::ReadReply]     Got 0
[SCPISocketTransport::SendCommand]     Sending CHAN4:SCALE?
[... 127 lines cut ...]
azonenberg/scopehal-apps#4  0x00007ffff66d1511 in  () at /usr/lib/libglib-2.0.so.0
azonenberg/scopehal-apps#5  0x00007ffff5f0c422 in start_thread () at /usr/lib/libpthread.so.0
azonenberg/scopehal-apps#6  0x00007ffff5e3bbf3 in clone () at /usr/lib/libc.so.6

#0  0x00007ffff5f16b0c in recv () at /usr/lib/libpthread.so.0
azonenberg/scopehal-apps#1  0x00007ffff640fe43 in Socket::RecvLooped(unsigned char*, int) (this=0x555555b6b748, buf=0x7fffdc001270 "\n", len=4000001)
    at /home/pepijn/code/scopehal-apps/lib/xptools/Socket.cpp:311
azonenberg/scopehal-apps#2  0x00007ffff63a303d in SCPISocketTransport::ReadRawData(unsigned long, unsigned char*) (this=
    0x555555b6b740, len=4000001, buf=0x7fffdc001270 "\n") at /home/pepijn/code/scopehal-apps/lib/scopehal/SCPISocketTransport.cpp:135
azonenberg/scopehal-apps#3  0x00007ffff63f4d77 in RigolOscilloscope::AcquireData(bool) (this=0x555555b26800, toQueue=true)
    at /home/pepijn/code/scopehal-apps/lib/scopehal/RigolOscilloscope.cpp:599
azonenberg/scopehal-apps#4  0x00005555556f0ec7 in ScopeThread(Oscilloscope*) (scope=0x555555b26800)
    at /home/pepijn/code/scopehal-apps/src/glscopeclient/main.cpp:358
azonenberg/scopehal-apps#5  0x00005555556afd7c in std::__invoke_impl<void, void (*)(Oscilloscope*), Oscilloscope*>(std::__invoke_other, void (*&&)(Oscilloscope*), Oscilloscope*&&) (__f=@0x555555c40b40: 0x5555556f0caf <ScopeThread(Oscilloscope*)>) at /usr/include/c++/10.1.0/bits/invoke.h:60
azonenberg/scopehal-apps#6  0x00005555556afcd5 in std::__invoke<void (*)(Oscilloscope*), Oscilloscope*>(void (*&&)(Oscilloscope*), Oscilloscope*&&)
    (__fn=@0x555555c40b40: 0x5555556f0caf <ScopeThread(Oscilloscope*)>) at /usr/include/c++/10.1.0/bits/invoke.h:95
azonenberg/scopehal-apps#7  0x00005555556afc45 in std::thread::_Invoker<std::tuple<void (*)(Oscilloscope*), Oscilloscope*> >::_M_invoke<0ul, 1ul>(std::_Index_tuple<0ul, 1ul>) (this=0x555555c40b38) at /usr/include/c++/10.1.0/thread:264
azonenberg/scopehal-apps#8  0x00005555556afbfe in std::thread::_Invoker<std::tuple<void (*)(Oscilloscope*), Oscilloscope*> >::operator()()
    (this=0x555555c40b38) at /usr/include/c++/10.1.0/thread:271
azonenberg/scopehal-apps#9  0x00005555556afbe2 in std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (*)(Oscilloscope*), Oscilloscope*> > >::_M_run() (this=0x555555c40b30) at /usr/include/c++/10.1.0/thread:215
azonenberg/scopehal-apps#10 0x00007ffff6196b74 in std::execute_native_thread_routine(void*) (__p=0x555555c40b30)
    at /build/gcc/src/gcc/libstdc++-v3/src/c++11/thread.cc:80
azonenberg/scopehal-apps#11 0x00007ffff5f0c422 in start_thread () at /usr/lib/libpthread.so.0
azonenberg/scopehal-apps#12 0x00007ffff5e3bbf3 in clone () at /usr/lib/libc.so.6
```

azonenberg, 2020-12-03:
New root cause identified. Reopened and migrated to scopehal repo.

bvernoux, 2020-12-03:
Feedback are welcome here it is rock stable with Windows7 64bits build with 0.94 WFM/s
[image]

azonenberg, 2020-12-03:
Hopefully fixed for good in 6bfbfb4.

closed 2020-12-03 by azonenberg