ngscopeclient/scopehal #361: USB Decoder does not function correctly.

Codysseus, 2020-12-03:
I wrote a simple test case for the USB decoder that the offset of a given packet plus the length did not exceed the offset of the next packet. However, when I tried this on my own data I found that this did not hold true.
[Here](https://web.engr.oregonstate.edu/~hollidac/usbcsv.cpp) is a link to the modified `usbcsv` example with the test, and [here](web.engr.oregonstate.edu/~hollidac/init.7z) is a link to the 7z that contains the csv in question. Be careful as the CSV is almost 2GB.

azonenberg, 2020-12-06:
Just checking whether the fixes I mentioned on IRC solved this?

Codysseus, 2020-12-07:
Hey Andrew! Thank you for following up with this. At the moment I do not have the bandwidth to debug, but I will get back to you on Tuesday (Dec 08 2020) with what I have tried. I hope you had a nice weekend!

Codysseus, 2020-12-15:
So one thing RE: my sample. Channel 0-Analog is D-, and Channel 1-Analog is D+. I was able to load a Low Speed sample into glscopeclient and it seemed to label the sample correctly. I flipped the channels and set it to Low Speed, now packets are emitted, but there is still an overlap issue. Here is a picture.

[image]

Codysseus, 2021-03-22:
Hello Andrew,
Just wanted to update this. I tested the decoder again, and the decoder works in the glscopeclient. It must be something to do with the configuration of the headless program.
Bug issue closed!

closed 2021-03-22 by Codysseus