ngscopeclient/scopehal #386: SCPITMCTransport: reading a reply via multiple ReadRawData() calls fails

azonenberg, 2020-12-17:
The current implementation will break any logic that uses multiple ReadRawData() calls to read a variable length SCPI response (length in one call then data in the next). At least Tek MSO6 is affected.

@tomverbeure I believe this was your code originally?

tomverbeure, 2020-12-17:
In general, my Siglent scope on USBTMC is very broken, requiring a power cycle of the scope as soon as something went wrong, so I wasn't able to do a whole lot of debugging on this.

That said, if this particular function is giving you issues on the Tek scope, chances are that you will have the same issue when using the Lxi driver, because most of the code is a copy.

azonenberg, 2020-12-17:
Interestingly no. I get buffer exhausted errors with USBTMC that render the scope unusable, while it (mostly) works via raw TCP or LXI.

tomverbeure, 2020-12-17:
Can you change [this line](https://github.com/azonenberg/scopehal/blob/b135693f9f955ea8e4a3471b6542cb0d7398083b/scopehal/SCPITMCTransport.cpp#L166) to `#if 1` and see what happens?

This is assuming that your Linux kernel has rev2 of the USBTMC driver installed.

azonenberg, 2021-05-12:
Moved to v0.2 as not currently critical and I don't have access to an MSO64 to test with.