ngscopeclient/scopehal #443: Heartbeat

mubes, 2021-05-01:
It would be useful to know that connectivity to the scope is good. This could be shown with a pulsing 'heartbeat' next to the number of WFMs at the bottom right of the UI panel. Multiple devices connected could have multiple HBs.

azonenberg, 2021-05-01:
As with #442, this requires talking to the scope constantly and it's very easy to introduce latency when we actually need to send traffic. Not calling this a wontfix at this point in time, but it needs to be carefully thought about.

azonenberg, 2021-05-01:
If the trigger is *armed* then it would be easy to add an activity indicator showing that poll messages are being received.

But if halted, less easy.