ngscopeclient/scopehal #494: CSV Import issue with floats (same comma both used as separator and decimal point).

enjoy-digital, 2021-06-03:
Hi,

first thanks for this awesome project! I wanted to explore it for a long time but haven't had the chance to do so before now.

While playing with the CSV import, I had difficulties importing float values and then noticed that comma `,` was both used as the CSV separator and float decimal point which then prevents import from working correctly with floats. By adapting the code to use a different separator I got the CSV import to work with floats.

So the decimal point should probably be changed to `.` or the CSV separator changed to something else.

azonenberg, 2021-06-03:
Did your generated data use commas as a decimal separator? If so, that's a data source problem and your files are malformed. If the code is expecting commas as decimal separators then we've got locale problems.

CSV by definition uses commas as field separators, and they're not legal in field content.

enjoy-digital, 2021-06-03:
I was initially generating the CSV using `,` as CSV separator and `.` as decimal point, here is a minimal repro:

CSV file:
```
Time, ADC0
0.0000000000, 2.500000
0.0000000242, 2.500000
0.0000000485, 3.500000
0.0000000727, 3.500000
```

With some debug added to `MockOscilloscope.cpp`:
```
diff --git a/scopehal/MockOscilloscope.cpp b/scopehal/MockOscilloscope.cpp
index 5b89729..ca99421 100644
--- a/scopehal/MockOscilloscope.cpp
+++ b/scopehal/MockOscilloscope.cpp
@@ -566,6 +566,7 @@ bool MockOscilloscope::LoadCSV(const string& path)
                        {
                                float f;
                                sscanf(tmp.c_str(), "%f", &f);
+                               printf("%f\n", f);
                                row.push_back(f);
 
                                if(line[i] == '\0')

```

We get the following when  importing the CSV:
```
0,000000
0,000000
2,000000
0,000000
2,000000
0,000000
3,000000
0,000000
3,000000
```

azonenberg, 2021-06-03:
Yeah, looks like the issue is printf/scanf not using . as the decimal separator here. Will look into it.

commit 3d74746b (2021-06-04): Unit class now uses the configured locale for printing numbers for display, keeping the thread's locale at "C" the rest of the time so we always use '.' as the decimal separator for file I/O. Fixes #494.
closed 2021-06-04 by azonenberg