ngscopeclient/scopehal #498: Pulse generator de-embedding filter

azonenberg, 2021-06-11:
Given a memory trace of a pulse generator's output, model it as an ideal unit step followed by an unknown channel and compute the S21 response of that channel (to be displayed or exported to a file).

someone--else, 2021-06-11:
In theory, it should also be possible to apply standard VNA calibration techniques (SOLT/TRL etc) to TDR/TDT traces as well, thus calibrating the step generator/scope frequency response details out. This would have the advantage of not assuming step to be ideal and also extending frequency response beyond step generator/scope -3db bandwidth (at the expense of some dynamic range)

someone--else, 2021-06-11:
(And the simplest version of TRL cal would be just assuming there are no reflections and measuring a through/source step signal and postulating that is a flat 0db S21)