ngscopeclient/scopehal #499: "CSV" from HP 'scope causes segfault

thirtythreeforty, 2021-06-14:
The attached file [WAV00.csv](https://github.com/azonenberg/scopehal-apps/files/6644880/WAV00.csv) causes a segfault when importing it:

```
#0  MockOscilloscope::NormalizeTimebases() (this=0x555555873800) at /home/georgev/Code/scopehal-apps/lib/scopehal/MockOscilloscope.cpp:1069
azonenberg/scopehal-apps#1  0x00007ffff64c1b89 in MockOscilloscope::LoadCSV(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
    (this=<optimized out>, path="/home/georgev/WAV00.csv") at /home/georgev/Code/scopehal-apps/lib/scopehal/MockOscilloscope.cpp:1051
azonenberg/scopehal-apps#2  0x00005555555fcc4d in OscilloscopeWindow::ImportCSVToNewSession(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (this=0x55555597d430, filename="/home/georgev/WAV00.csv") at /home/georgev/Code/scopehal-apps/src/glscopeclient/OscilloscopeWindow.cpp:964
azonenberg/scopehal-apps#3  0x00005555555fdc48 in OscilloscopeWindow::OnFileImport() (this=0x7fffffffce90)
    at /home/georgev/Code/scopehal-apps/src/glscopeclient/OscilloscopeWindow.cpp:805
azonenberg/scopehal-apps#4  0x00007ffff6c1ce49 in Glib::SignalProxyNormal::slot0_void_callback(_GObject*, void*) () at /usr/lib/libglibmm-2.4.so.1
azonenberg/scopehal-apps#5  0x00007ffff68a1ddf in g_closure_invoke () at /usr/lib/libgobject-2.0.so.0
azonenberg/scopehal-apps#6  0x00007ffff68cb1d8 in  () at /usr/lib/libgobject-2.0.so.0
azonenberg/scopehal-apps#7  0x00007ffff68becad in g_signal_emit_valist () at /usr/lib/libgobject-2.0.so.0
azonenberg/scopehal-apps#8  0x00007ffff68bf210 in g_signal_emit () at /usr/lib/libgobject-2.0.so.0
azonenberg/scopehal-apps#9  0x00007ffff72ac2a5 in gtk_widget_activate () at /usr/lib/libgtk-3.so.0
azonenberg/scopehal-apps#10 0x00007ffff717245d in gtk_menu_shell_activate_item () at /usr/lib/libgtk-3.so.0
azonenberg/scopehal-apps#11 0x00007ffff7172734 in  () at /usr/lib/libgtk-3.so.0
azonenberg/scopehal-apps#12 0x00007ffff70041e5 in  () at /usr/lib/libgtk-3.so.0
azonenberg/scopehal-apps#13 0x00007ffff68bf096 in g_signal_emit_valist () at /usr/lib/libgobject-2.0.so.0
azonenberg/scopehal-apps#14 0x00007ffff68bf210 in g_signal_emit () at /usr/lib/libgobject-2.0.so.0
azonenberg/scopehal-apps#15 0x00007ffff72bf6f5 in  () at /usr/lib/libgtk-3.so.0
azonenberg/scopehal-apps#16 0x00007ffff715d3b5 in  () at /usr/lib/libgtk-3.so.0
azonenberg/scopehal-apps#17 0x00007ffff715e2d3 in gtk_main_do_event () at /usr/lib/libgtk-3.so.0
azonenberg/scopehal-apps#18 0x00007ffff6ebe7f3 in  () at /usr/lib/libgdk-3.so.0
azonenberg/scopehal-apps#19 0x00007ffff6f0e554 in  () at /usr/lib/libgdk-3.so.0
azonenberg/scopehal-apps#20 0x00007ffff67ae02c in g_main_context_dispatch () at /usr/lib/libglib-2.0.so.0
azonenberg/scopehal-apps#21 0x00007ffff6801b59 in  () at /usr/lib/libglib-2.0.so.0
azonenberg/scopehal-apps#22 0x00007ffff67ab781 in g_main_context_iteration () at /usr/lib/libglib-2.0.so.0
azonenberg/scopehal-apps#23 0x00007ffff7154353 in gtk_main_iteration_do () at /usr/lib/libgtk-3.so.0
azonenberg/scopehal-apps#24 0x00007ffff7c183be in Gtk::Main::iteration(bool) () at /usr/lib/libgtkmm-3.0.so.1
azonenberg/scopehal-apps#25 0x00005555556343a2 in ScopeApp::run(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, bool, bool, bool, bool, bool)
    (this=this@entry=0x5555556bfb90, filesToLoad=std::vector of length 0, capacity 0, reconnect=reconnect@entry=false, nodata=nodata@entry=false, retrigger=retrigger@entry=false, nodigital=nodigital@entry=false, nospectrum=false)
    at /home/georgev/Code/scopehal-apps/src/glscopeclient/ScopeApp.cpp:143
azonenberg/scopehal-apps#26 0x00005555555bf86d in main(int, char**) (argc=<optimized out>, argv=<optimized out>)
    at /home/georgev/Code/scopehal-apps/src/glscopeclient/main.cpp:215
```

This was exported as "YValues" from my scope.

I would expect it to import with some arbitrary or user-specified timebase.

azonenberg, 2021-06-14:
So this file provides Y values only, no timebase? Interesting. This definitely violates our input format spec but shouldn't crash.

Does your scope support exporting both X and Y values? IMO the correct way to handle this file is just to report an error and tell the user to submit something well formed.

thirtythreeforty, 2021-06-14:
Yes, XYValues is another option. It's also trivial to backfill the timebase
in Excel.

Reason I picked Y-only is because it takes a nontrivial amount of time to
write 400KB to a floppy disk... if that gives you an idea of the age of the
thing, haha

azonenberg, 2021-06-14:
I just pushed a commit that should result in MockOscilloscope::LoadCSV() returning an error and failing, rather than segfaulting, when presented with a CSV that has only one column. From my perspective the bug (the crash) is now fixed.

I intend to close this issue as soon as you confirm this fixes your problem; if you want to file a separate lower priority enhancement request for allowing import of Y-only CSV files then feel free. But I probably won't get to it any time soon as my focus is getting ready for the v0.1 release.

thirtythreeforty, 2021-06-14:
Great, thanks for looking! Confirmed no longer segfaulting. It's not a super specific error...

[image]

but agree, I think UI improvements would be great but not high priority.

closed 2021-06-14 by thirtythreeforty

azonenberg, 2021-06-14:
There's a more detailed message on stdout. The current MockOscilloscope API only provides a success/fail status to the caller so the detailed info has to be printed elsewhere. We can clean this up in the future.