ngscopeclient/scopehal #502: Windows/MINGW64 build is broken

MegabytePhreak, 2021-06-17:
I'm trying to build using the instructions from the manual, but I encountered some errors. 

First error, which was able to resolve by adding .c_str():
```
C:/Users/Megabytephreak/Code/scopehal-apps/lib/scopehal/Unit.cpp: In static member function 'static void Unit::SetPrintingLocale()':
C:/Users/Megabytephreak/Code/scopehal-apps/lib/scopehal/Unit.cpp:382:25: error: cannot convert 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'const char*'
  382 |   setlocale(LC_NUMERIC, m_slocale);
      |                         ^~~~~~~~~
      |                         |
      |                         std::string {aka std::__cxx11::basic_string<char>}
In file included from C:/msys64/mingw64/include/c++/10.3.0/clocale:42,
                 from C:/msys64/mingw64/include/c++/10.3.0/x86_64-w64-mingw32/bits/c++locale.h:41,
                 from C:/msys64/mingw64/include/c++/10.3.0/bits/localefwd.h:40,
                 from C:/msys64/mingw64/include/c++/10.3.0/string:43,
                 from C:/Users/Megabytephreak/Code/scopehal-apps/lib/scopehal/scopehal.h:42,
                 from C:/Users/Megabytephreak/Code/scopehal-apps/lib/scopehal/Unit.cpp:30:
C:/msys64/mingw64/x86_64-w64-mingw32/include/locale.h:90:53: note:   initializing argument 2 of 'char* setlocale(int, const char*)'
   90 |   char *__cdecl setlocale(int _Category,const char *_Locale);
      |                                         ~~~~~~~~~~~~^~~~~~~
make[2]: *** [lib/scopehal/CMakeFiles/scopehal.dir/build.make:118: lib/scopehal/CMakeFiles/scopehal.dir/Unit.cpp.obj] Error 1
```

Second error, which I've not yet resolved:

```
[ 29%] Linking CXX shared library libscopeprotocols.dll
cd /C/Users/Megabytephreak/Code/scopehal-apps/build/lib/scopeprotocols && /C/msys64/mingw64/bin/cmake.exe -E rm -f CMakeFiles/scopeprotocols.dir/objects.a
cd /C/Users/Megabytephreak/Code/scopehal-apps/build/lib/scopeprotocols && /C/msys64/mingw64/bin/ar.exe cr CMakeFiles/scopeprotocols.dir/objects.a @CMakeFiles/scopeprotocols.dir/objects1.rsp
cd /C/Users/Megabytephreak/Code/scopehal-apps/build/lib/scopeprotocols && /C/msys64/mingw64/bin/g++.exe -g -Wall -Wextra -Wuninitialized  -Wshadow -Wpedantic -Wcast-align -Wwrite-strings -Wmissing-declarations -Wvla -Wunsafe-loop-optimizations -mtune=native -ffast-math -D_USE_MATH_DEFINES -D_POSIX_THREAD_SAFE_FUNCTIONS -fopenmp -O3 -pipe -Wl,--dynamicbase,--high-entropy-va,--nxcompat,--default-image-base-high -shared -o libscopeprotocols.dll -Wl,--out-implib,libscopeprotocols.dll.a -Wl,--major-image-version,0,--minor-image-version,0 -Wl,--whole-archive CMakeFiles/scopeprotocols.dir/objects.a -Wl,--no-whole-archive  ../scopehal/libscopehal.dll.a /C/msys64/mingw64/lib/libffts.dll.a ../xptools/libxptools.a -lws2_32 ../log/liblog.dll.a ../graphwidget/libgraphwidget.dll.a -lsigc-2.0 -lgtkmm-3.0 -latkmm-1.6 -lgdkmm-3.0 -lgiomm-2.4 -lgtk-3 -lgdk-3 -lz -lgdi32 -limm32 -lshell32 -lole32 -lwinmm -ldwmapi -lsetupapi -lcfgmgr32 -lpangowin32-1.0 -latk-1.0 -lcairo-gobject -lgio-2.0 -lpangomm-1.4 -lglibmm-2.4 -lcairomm-1.0 -lsigc-2.0 -lgtkmm-3.0 -latkmm-1.6 -lgdkmm-3.0 -lgiomm-2.4 -lgtk-3 -lgdk-3 -lz -lgdi32 -limm32 -lshell32 -lole32 -lwinmm -ldwmapi -lsetupapi -lcfgmgr32 -lpangowin32-1.0 -latk-1.0 -lcairo-gobject -lgio-2.0 -lpangomm-1.4 -lglibmm-2.4 -lcairomm-1.0 -lpangocairo-1.0 -lpango-1.0 -lharfbuzz -lcairo -lgdk_pixbuf-2.0 -lgobject-2.0 -lglib-2.0 -lintl /C/msys64/mingw64/lib/libyaml-cpp.a -lshlwapi /C/msys64/mingw64/lib/libffts.dll.a /C/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/10.3.0/libgomp.dll.a /C/msys64/mingw64/x86_64-w64-mingw32/lib/libmingwthrd.a -lkernel32 -luser32 -lgdi32 -lwinspool -lshell32 -lole32 -loleaut32 -luuid -lcomdlg32 -ladvapi32
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/msys64/mingw64/lib/libyaml-cpp.a(exceptions.cpp.obj):(.text+0x180): multiple definition of `YAML::InvalidNode::~InvalidNode()'; ../scopehal/libscopehal.dll.a(d001516.o):(.text+0x0): first defined here
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/msys64/mingw64/lib/libyaml-cpp.a(exceptions.cpp.obj):(.text+0x1c0): multiple definition of `YAML::BadConversion::~BadConversion()'; ../scopehal/libscopehal.dll.a(d001528.o):(.text+0x0): first defined here
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/msys64/mingw64/lib/libyaml-cpp.a(exceptions.cpp.obj):(.text+0x240): multiple definition of `YAML::BadSubscript::~BadSubscript()'; ../scopehal/libscopehal.dll.a(d001523.o):(.text+0x0): first defined here
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/msys64/mingw64/lib/libyaml-cpp.a(node_data.cpp.obj):(.text+0x3a0): multiple definition of `YAML::detail::node_data::mark_defined()'; ../scopehal/libscopehal.dll.a(d001574.o):(.text+0x0): first defined here
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/msys64/mingw64/lib/libyaml-cpp.a(node_data.cpp.obj):(.text+0x4f0): multiple definition of `YAML::detail::node_data::set_null()'; ../scopehal/libscopehal.dll.a(d001587.o):(.text+0x0): first defined here
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/msys64/mingw64/lib/libyaml-cpp.a(node_data.cpp.obj):(.text+0x500): multiple definition of `YAML::detail::node_data::set_scalar(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'; ../scopehal/libscopehal.dll.a(d001572.o):(.text+0x0): first defined here
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/msys64/mingw64/lib/libyaml-cpp.a(node_data.cpp.obj):(.text+0x740): multiple definition of `YAML::detail::node_data::begin()'; ../scopehal/libscopehal.dll.a(d001582.o):(.text+0x0): first defined here
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/msys64/mingw64/lib/libyaml-cpp.a(node_data.cpp.obj):(.text+0x880): multiple definition of `YAML::detail::node_data::end()'; ../scopehal/libscopehal.dll.a(d001579.o):(.text+0x0): first defined here
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/msys64/mingw64/lib/libyaml-cpp.a(node_data.cpp.obj):(.text+0xb00): multiple definition of `YAML::detail::node_data::insert_map_pair(YAML::detail::node&, YAML::detail::node&)'; ../scopehal/libscopehal.dll.a(d001577.o):(.text+0x0): first defined here
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/msys64/mingw64/lib/libyaml-cpp.a(node_data.cpp.obj):(.text+0x12c0): multiple definition of `YAML::detail::node_data::convert_to_map(std::shared_ptr<YAML::detail::memory_holder>)'; ../scopehal/libscopehal.dll.a(d001575.o):(.text+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
```

Seems like there is an issue with how yaml-cpp is being linked. Perhaps something different about how DLLs vs SOs work. 

I know windows support is experimental, but do you know the last time it work working so I can try to bisect?

azonenberg, 2021-06-17:
We've known about the linker issue for probably a week now, CI complains every build. I don't have a Windows machine to troubleshoot so I'm at the mercy of our handful of Windows devs, none of whom have stepped up to fix it. Check history of CMakeLists.txt for the last few commits and you'll probably find the offending change?

As far as the first problem, good catch. Once the build is broken, further breakage is easy to miss.

azonenberg, 2021-06-17:
AFAIK the ultimate problem is that yaml-cpp ends up being linked twice on Windows and you get multiply defined symbols. From the error message it seems that it's trying to pull in libyaml-cpp.a which is wrong (this should be dynamically linked, not statically.

MegabytePhreak, 2021-06-17:
Agreed on the issue. 
I think this might have to do with PR #489, but I'll try to bisect.

MegabytePhreak, 2021-06-17:
In the first broken CI build, where `find_package(Yaml REQUIRED)` gets added, the definition of YAML_LIBRARIES uses a static library on MINGW, but a dynamic library on Linux
https://github.com/azonenberg/scopehal-apps/runs/2713477439#step:5:250
https://github.com/azonenberg/scopehal-apps/runs/2713477350#step:6:61

Using the old yaml-cpp in place of ${YAML_LIBRARIES} does allow the build to progress further, but there's some string/wstring breakage in PreferenceManager.cpp I need to resolve next.

azonenberg, 2021-06-19:
Ok, I think all of the build failures are fixed finally. Thanks!

closed 2021-06-19 by azonenberg