ngscopeclient/scopehal #507: Rigol swapping channels and invalid data every 4096 samples

x44203, 2021-06-26:
[image]
For example in the attached image channels 2 and 3 are invalid and channels 1 and 4 get swapped. This is with a build from the latest main branch at the time of creation of this post. In the example the second spike has gone low at 4032 µs (8064 samples, with sample 0 at 0 fs) and starts going high at 4095.5 µs (8191 samples) So essentially for 128 samples it bugs out, with 4096 - 128 samples of space inbetween.

azonenberg, 2023-03-09:
@x44203 can you please test this with a recent build and see if it's fixed yet?

azonenberg, 2023-11-03:
@hansemro have you seen this / fixed it in your recent fixes? If not, can we close this?

hansemro, 2023-11-03:
I have not encountered/fixed this.

azonenberg, 2023-11-03:
Well I asked if it was still happening in the current version in March and haven't heard back. If nobody else has hit it let's close.

closed 2023-11-03 by azonenberg