ngscopeclient/scopehal #528: TouchstoneParser fails to load files with \r\n line endings on Windows

TerryHoCQ, 2021-12-26:
Merry Christmas! I'm trying to generate TDR waveform from a S2p file (On offline mode), and  I found 2 issues on importing touchstone file, thank you!

1: Fail to import touchstone from File menu
2: Can import touchstone file from Add/import menu, but no waveform generated.

azonenberg, 2021-12-26:
You seem to be zoomed really far out on the TDR plot.

For 50 GHz S-parameters you're going to want your stimulus waveform to be a minimum of 100 Gsps (minimum Nyquist rate), and the curves will look more smooth if you go even higher. I'd suggest 500 Gsps.

Here's what I get when zooming in on test.s2p S11 applied to a 500 Gsps step. This looks plausible for a transmission line with a nominal impedance of about 52 ohms plus a bit of loss causing an upward slope to the curve. The overall reflection curve is about 900ps long, so your line is about 450ps propagation delay one way (which lines up nicely with the group delay of your .s2p). Assuming around 6ps / mm propagation velocity your line is probably about 75mm in physical length.

[image]

azonenberg, 2021-12-26:
Also, try changing the fopen() call in TouchstoneParser::Load() (line 60 of lib/scopehal/TouchstoneParser.cpp) to open the file in binary mode ("rb" instead of "r"). Does that fix the import issue? Opening in text mode *should* work fine on Windows, but who knows...

TerryHoCQ, 2021-12-26:
> Also, try changing the fopen() call in TouchstoneParser::Load() (line 60 of lib/scopehal/TouchstoneParser.cpp) to open the file in binary mode ("rb" instead of "r"). Does that fix the import issue? Opening in text mode _should_ work fine on Windows, but who knows...

I have rebuilt the code, and confirmed it works fine by changingTouchstoneParser::Load() ("rb" instead of "r") on Windows.

TerryHoCQ, 2021-12-26:
And I tried to set the step rate with 500 Gsps, it seems the Channel emulation waveform is not correct. Thank you
[image]
!

azonenberg, 2021-12-26:
I'm not sure it's actually wrong. You're zoomed waaaaaay out and seem to be seeing low frequency / interpolation artifacts caused by the limited low frequency content of your .s2p. Try zooming in the channel emulation view to +/- 2ns of the step or so.

I'm closing this issue as the original line ending issue is now fixed and I don't want to mix different problems in one ticket. Feel free to file a second one if you're having further trouble.

commit 2f681b7c (2021-12-26): TouchstoneParser: open Touchstone files in binary mode on Windows so that files with Unix line endings are correctly interpreted. Fixes #528.
closed 2021-12-26 by azonenberg