ngscopeclient/scopehal #547: General hardware recommendation

ozel, 2022-02-04:
it seems the project reached a level of sophistication and general enthusiasm, that a brief shortlist of oscilloscope recommendations would be really nice. 
Listing 2-3 device models/series that are currently most feature complete together with glscopeclient.
I'd love to compare my future purchase plans with such a list and guess others would too.

azonenberg, 2022-02-04:
Section 5 of the manual (draft copy at https://www.antikernel.net/temp/glscopeclient-manual.pdf but we're still working on it) discusses the instrument drivers and known working models for each.

Generally speaking, Teledyne LeCroy is our "flagship" platform - it's what I use in my lab so I run it on a daily basis, their scopes are reasonably fast over SCPI so it performs well, and it's basically the reference driver that I test new features in before adding them elsewhere. They have an extremely stable API/command set so pretty much any of their scopes from the Windows 2000 era to current generation should work out of the box.

PicoScope 6000E series has excellent streaming performance - actually more WFM/s than the LeCroy platforms - but the driver is still experimental and is missing a few features. Other PicoScope families are in the works, but less stable. Pico is a big supporter of the project and provided me with a 6824E as a dev unit.

Siglent SDS2000x series is a bit slow due to the low-end hardware, but is probably the best of the budget platforms and is reasonably well supported. The 1000 series is supported but that driver is brand-new and likely still rough around the edges. The 2000 series driver probably works on the higher end Siglent platforms as well but has never been tested on them. Siglent is also supportive of the project and has been working with us on fixing firmware issues and improving performance.

This is by no means an all-inclusive list of supported models, see the manual for that. 

Additionally, we've been compiling a spreadsheet of performance figures. This data includes the scope make and model (we should probably start logging firmware version too), number of enabled channels, memory depth, and the number of waveforms per second we got. From this, the spreadsheet calculates the throughput in both samples/sec and bits/sec.

You can check out the spreadsheet at https://docs.google.com/spreadsheets/d/1Aw3pIdjKqfiHG1SAWSzkXyH3VJw_Mj2g_8I6pCXu0nY/edit#gid=0. This is again not an all-inclusive list of supported models, but it gives you an idea of what performance to expect from some popular options.

ozel, 2022-02-06:
thanks a lot for taking the time to respond in such detail! This was the guidance I was looking for.
I've of course seen chapter 5, but was more curious about features than speed comparison - knowing that at least some older scopes do not expose a lot of remote control functionality or sometimes have a buggy API.
LeCroy offers currently a big promotion discount in Europe for the Wavesurfer 4000HD series which I will checkout.