ngscopeclient/scopehal #549: Raw complex I/Q export

azonenberg, 2022-02-05:


dkozel, 2022-02-10:
As a thought, the SigMF metadata format might be useful to you here. It's specifically designed for allowing a raw datafile to be stored with a separate simple metadatafile alongside describing the properties of the recording. So no interference with inport/export with software that already/only supports raw files, no header or etc to mess with parsing and updating while loading/writing captures, you can use optimized/specific code to handle the data files and leave the metadata to the side. It also specifically handles multiple simultaneous streams, segmented captures, and signal annotations.

https://sigmf.org

If you do think this is interesting and have any questions I can possibly answer or link you with the spec authors.

777arc, 2022-02-11:
yeah would be sweet if SigMF was supported, and should be very little extra work needed since the data file is just a binary file anyway