ngscopeclient/scopehal #652: Siglent: Using SDS1104X-E through usbtmc not always work

racerxdl, 2022-07-06:
As discussed here (https://twitter.com/azonenberg/status/1544476705842049024) using a SDS1104X-E through usbtmc mostly dont work (glscope client correctly detects, and _sometimes_ the trace appears on it).

So I made some usb captures, logging and everything to try to help debugging the issue. Funny enough, through ethernet everything seens to work fine (I have some issues setting up the triggers correctly, but I'm not sure is a communication issue). I always though it might be a linux usbtmc driver issue since sigrok _sort_ of works correctly, but it does by using libusb to connecting directly to the oscilloscope instead using usbtmc.

Here are some logs from initialization. The traces do appear once in the glscopeclient, but then everything stops working and I need to remove the usb cable and put again so I can get the trace again (just restarting glscopeclient does not help)

[usbcapture_filtered.pcapng.gz](https://github.com/glscopeclient/scopehal/files/9050517/usbcapture_filtered.pcapng.gz)
[dmesg.log](https://github.com/glscopeclient/scopehal/files/9050520/dmesg.log)
[sds1104x-e.log](https://github.com/glscopeclient/scopehal/files/9050521/sds1104x-e.log)

I'm currently on x86-64 ubuntu 22.04 (kernel 5.15.0). Once I managed to completely freeze the osciloscope trying to capture from glscopeclient (I had to hardreset the osciloscope to get back), but I really dont remember what I did to cause that.

My oscilloscope is a Siglent SDS1104X-E key-upgraded to SDS1204X-E running latest firmware (8.1.6.1.37R9)

mandl, 2023-01-04:
Hello racerxdl,
i did some debugging today. It looks like a  firmware bug. The maximum request size is 48 byte.
I am running the latest firmware 8.2.6.1.37R9 on
`5.15.0-56-generic #62-Ubuntu SMP Tue Nov 22 19:54:14 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux`

Please take a look at siglent.patch for SCPITMCTransport.cpp

```
do
{
  // workaround for Siglent SDS1x04X-E   max request size is 48 ?
  // Bug in firmware 8.2.6.1.37R9 ?
  bytes_requested = 48 ; 
  bytes_fetched = read(m_handle, (char *)m_staging_buf + i, 48);
  i += bytes_fetched;
} while(bytes_fetched == bytes_requested);
```

[siglent.patch.txt](https://github.com/glscopeclient/scopehal/files/10345806/siglent.patch.txt)

[image]

modinfo usbtmc

filename:       /lib/modules/5.15.0-56-generic/kernel/drivers/usb/class/usbtmc.ko
license:        GPL
srcversion:     A55300AFA898251D26323BB
alias:          usb:v*p*d*dc*dsc*dp*icFEisc03ip01in*
alias:          usb:v*p*d*dc*dsc*dp*icFEisc03ip00in*
depends:        
retpoline:      Y
intree:         Y
name:           usbtmc
vermagic:       5.15.0-56-generic SMP mod_unload modversions

mandl, 2023-01-04:
**Do some debugging**
Unload the usbtmc  module
`rmmod usbtmc`

Load module with debug enable
`insmod usbtmc.ko dyndpg=+p`

Run 
`glscopeclient --debug --trace SCPITMCTransport`

`dmesg`

azonenberg, 2023-01-04:
Interesting but not surprising. This is something we should probably add as an additional argument to the usbtmc driver (after the /dev path) since enabling the workaround all the time will have serious performance impacts for non-buggy instruments.

mandl, 2023-01-06:
Hello Andrew,
now we can configure the transfer size.

for example: opening /dev/ubtmc0 with transfer size 48 

/dev/usbtmc0:48

or

glscopeclient myscope:siglent:usbtmc:/dev/usbtmc0:48

[image]

azonenberg, 2023-01-17:
Merged the fix to scopehal. Ready to close once @racerxdl confirms the patch fixes it.

racerxdl, 2023-01-19:
Hi @mandl / @azonenberg with that with the scopehal fix and opening with the :48 option it does work without any issues.

closed 2023-01-19 by azonenberg