ngscopeclient/scopehal #660: Look into replacing FFTS calls with vkFFT

azonenberg, 2022-07-24:
FFTS is old and not as fast as I'd like.

We're moving towards Vulkan compute being a fundamental requirement for the platform, so it might be possible to replace *all* of our FFTs with it. We need to think about how this will work in the ATE use case though: can vulkan compute run on integrated GPUs / headless platforms? Will we need to put a discrete GPU in an ATE system?

azonenberg, 2024-01-29:
The only remaining use of FFTS is DegradeSerialData() in TestWaveformSource.

azonenberg, 2024-04-12:
Are you on x86 or ARM?

At one point, not sure if fixed, we had a bug where FFTS was disabled on apple silicon builds for macos but not for x86 builds.

FFTS is currently only used for the demo oscilloscope and some unit tests. It's not far from being completely gone but I've been too busy to completely remove it. If you feel adventurous and want to poke around the code, building without tests (I think this is default) and disabling DemoOscilloscope should get you a working binary in the meantime.

sslupsky, 2024-04-12:
At the moment attempting to install on x86 (but I do have an M2 Mac I would also like to install it on).

Ok, I'll poke around a bit.  I already tried eliminating it from the cmake file but the build failed because it was not included.

azonenberg, 2024-04-12:
Actually, not even the whole demo scope. Just making TestWaveformSource::DegradeSerialData be a no-op would probably be sufficient.

sslupsky, 2024-04-12:
Seems to also be used in "Filter_FFT.cpp" as well.

Appears to be building ...

azonenberg, 2024-04-12:
> Seems to also be used in "Filter_FFT.cpp" as well.
> 
> Appears to be building ...

That's one of the unit tests. You should be able to turn off all of the tests with a CMake option.

sslupsky, 2024-04-12:
It built with no errors.  When I run the app, it throws an error.  Not sure if this is a problem with my Homebrew installation of Vulkan or if there is some other issue.  There was no app bundle, just a "ngscopeclient" executable.  Is that the normal build of ngscopeclient for macOS?  Or, is there another step to build the app bundle that I missed?

```
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[BronzeMtlRenderCmdEncoder setVertexBuffer:offset:attributeStride:atIndex:]: unrecognized selector sent to instance 0x7fcc1f8abc00'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007ff81607d2ee __exceptionPreprocess + 242
	1   libobjc.A.dylib                     0x00007ff815b6e21b objc_exception_throw + 48
	2   CoreFoundation                      0x00007ff81612265d -[NSObject(NSObject) __retain_OA] + 0
	3   libMoltenVK.dylib                   0x0000000181252f75 _ZNSt3__110__function6__funcIZN39MVKGraphicsResourcesCommandEncoderState10encodeImplEjE3$_6NS_9allocatorIS3_EEFvP17MVKCommandEncoderR19MVKMTLBufferBindingEEclEOS7_S9_ + 69
	4   libMoltenVK.dylib                   0x000000018124f704 _ZN39MVKGraphicsResourcesCommandEncoderState14encodeBindingsE14MVKShaderStagePKcbNSt3__18functionIFvP17MVKCommandEncoderR19MVKMTLBufferBindingEEENS4_IFvS6_S8_11MVKArrayRefIKjEEEENS4_IFvS6_R20MVKM	5   libMoltenVK.dylib                   0x0000000181250110 _ZN39MVKGraphicsResourcesCommandEncoderState10encodeImplEj + 272
	6   libMoltenVK.dylib                   0x00000001811bd4ec _ZN17MVKCommandEncoder17finalizeDrawStateE16MVKGraphicsStage + 76
	7   libMoltenVK.dylib                   0x00000001811b8193 _ZN17MVKCmdDrawIndexed6encodeEP17MVKCommandEncoder + 595
	8   libMoltenVK.dylib                   0x00000001811bb0ca _ZN17MVKCommandEncoder6encodeEPU27objcproto16MTLCommandBuffer11objc_objectP25MVKCommandEncodingContext + 250
	9   libMoltenVK.dylib                   0x00000001811bb52c _ZN16MVKCommandBuffer6submitEP31MVKQueueCommandBufferSubmissionP25MVKCommandEncodingContext + 236
	10  libMoltenVK.dylib                   0x0000000181229676 _ZN35MVKQueueFullCommandBufferSubmissionILm1EE20submitCommandBuffersEv + 102
	11  libMoltenVK.dylib                   0x0000000181227d76 _ZN31MVKQueueCommandBufferSubmission7executeEv + 198
	12  libMoltenVK.dylib                   0x00000001812261a9 _ZN8MVKQueue6submitI12VkSubmitInfoEE8VkResultjPKT_P9VkFence_T13MVKCommandUse + 329
	13  libMoltenVK.dylib                   0x0000000181171e7a vkQueueSubmit + 74
	14  ngscopeclient                       0x000000010e183b46 _ZN12VulkanWindow6RenderEv + 2742
	15  ngscopeclient                       0x000000010e1cf842 main + 850
	16  dyld                                0x00007ff815ba2366 start + 1942
)
libc++abi: terminating due to uncaught exception of type NSException
zsh: abort      ./ngscopeclient
```

azonenberg, 2024-08-30:
All use of FFTS outside of unit tests is now gone as of 1a704437bf094bbcdfa214c1e5d1f0778fed149a. Closing this ticket (finally!)

closed 2024-08-30 by azonenberg