ngscopeclient/scopehal #688: "Unrecognized dataset type" when importing Tek .WFM file

dan-gies, 2022-09-12:
## System Info:
Glscopeclient version: 0.1-0ec5f20
OS: Windows 10
Oscilloscope: Tektronix MSO56B (offline, open from .WFM file)

## Description:
A waveform (~900 MB) was captured from a Tektronix MSO56B. When trying to import the .WFM file, I get the following message: `ERROR: Unrecognized dataset type`. First 10k lines of .WFM file are attached.

[image]
[tek-10klines-wfm.txt](https://github.com/glscopeclient/scopehal-apps/files/9550278/tek-10klines-wfm.txt)

azonenberg, 2022-09-12:
Looks like this is an issue in WFMImportFilter. I added a bit of additional debug logging.

Please pull latest code from git (make sure you pull recursively to get submodule updates), build it, run glscopeclient with --debug --logfile log.txt, and attach the generated log.

dan-gies, 2022-09-15:
@azonenberg here you go! After attempting a single import of the file, this is what I see:

### Console:
```
Reading WFM file C:\Users\gies\Desktop\temp\Tek000_000.wfm
    Waveform version:     ":WFM#003"
    Digit count:          10
    File size:            1000000959 bytes
    Bytes per point:      2
    Curve data offset:    838 bytes
    Waveform label:
    Curve objects:        0
    Waveform header size: 2
    ERROR: Unrecognized dataset type 16777216
Reading WFM file C:\Users\gies\Desktop\temp\Tek000_000.wfm
    Waveform version:     ":WFM#003"
    Digit count:          10
    File size:            1000000959 bytes
    Bytes per point:      2
    Curve data offset:    838 bytes
    Waveform label:
    Curve objects:        0
    Waveform header size: 2
    ERROR: Unrecognized dataset type 16777216
```

### log.txt:
```
Warning: glscopeclient works best with the OMP_WAIT_POLICY environment variable set to PASSIVE
Initializing Vulkan
    VK_KHR_get_physical_device_properties2: supported
    VK_EXT_debug_utils: supported
    Loader/API support available for Vulkan 1.2
    Vulkan 1.2 support available, requesting it
    Skipping GLFW init to work around gtk gl/vulkan interop bug
    Physical devices:
        Device 0: NVIDIA GeForce RTX 2070 SUPER
            API version:            0x004020af (0.1.2.175)
            Driver version:         0x76030000 (472.12.0.0)
            Vendor ID:              10de
            Device ID:              1e84
            Device type:            Discrete GPU
            int64:                  yes
            int16:                  yes (allowed in SSBOs)
            int8:                   yes (allowed in SSBOs)
            Max image dim 2D:       32768
            Max storage buf range:  4095 MB
            Max mem alloc:          0 MB
            Max compute shared mem: 48 KB
            Max compute grp count:  2147483647 x 65535 x 65535
            Max compute invocs:     1024
            Max compute grp size:   1024 x 1024 x 64
            Memory types:
                Type 0
                    Heap index: 1
                Type 1
                    Heap index: 0
                    Device local
                Type 2
                    Heap index: 1
                    Host visible
                    Host coherent
                Type 3
                    Heap index: 1
                    Host visible
                    Host coherent
                    Host cached
                Type 4
                    Heap index: 2
                    Device local
                    Host visible
                    Host coherent
            Memory heaps:
                Heap 0
                    Size: 7 GB
                    Device local
                Heap 1
                    Size: 31 GB
                Heap 2
                    Size: 214 MB
                    Device local
        Selected device 0
            Queue families (3 total)
                Queue type 0
                    Queue count:          16
                    Timestamp valid bits: 64
                    Graphics
                    Compute
                    Transfer
                    Sparse binding
                    Using this queue type for compute
                Queue type 1
                    Queue count:          2
                    Timestamp valid bits: 64
                    Transfer
                    Sparse binding
                Queue type 2
                    Queue count:          8
                    Timestamp valid bits: 64
                    Compute
                    Transfer
                    Sparse binding
            Enabling 64-bit integer support
            Enabling 16-bit integer support
            Enabling 16-bit integer support for SSBOs
            Enabling 8-bit integer support
            Enabling 8-bit integer support for SSBOs
            Using type 3 for pinned host memory
            Using type 1 for card-local memory
    
    vkFFT version: 1.2.29
Detecting CPU features...
    * AVX2
    * FMA
    
    Warning: AVX2/AVX512 detected but disabled on MinGW64/GCC (see https://github.com/azonenberg/scopehal-apps/issues/29
    5)
Reading WFM file C:\Users\gies\Desktop\temp\Tek000_000.wfm
    Waveform version:     ":WFM#003"
    Digit count:          10
    File size:            1000000959 bytes
    Bytes per point:      2
    Curve data offset:    838 bytes
    Waveform label:       
    Curve objects:        0
    Waveform header size: 2
    ERROR: Unrecognized dataset type 16777216
Reading WFM file C:\Users\gies\Desktop\temp\Tek000_000.wfm
    Waveform version:     ":WFM#003"
    Digit count:          10
    File size:            1000000959 bytes
    Bytes per point:      2
    Curve data offset:    838 bytes
    Waveform label:       
    Curve objects:        0
    Waveform header size: 2
    ERROR: Unrecognized dataset type 16777216
```

azonenberg, 2022-09-15:
Should be fixed in https://github.com/glscopeclient/scopehal/commit/b1c9b639a53d25aba56d0e4ea649aa3823d966aa.

Please pull latest and verify.

azonenberg, 2022-10-04:
Presumed fixed, closing.

closed 2022-10-04 by azonenberg

dan-gies, 2022-10-05:
@azonenberg sorry for the delay. Confirmed: this is fixed.