ngscopeclient/scopehal #705: Segmentation Fault when adjusting waveform sizing

dan-gies, 2022-10-05:
## System Info:
Glscopeclient version: 0.1-59ef026
OS: Windows 10
Oscilloscope: SDS6204A

## Description:
While the scope was continuously triggering, I adjusted the Ch1 waveform vertical sizing with my mouse (grab the bottom sizing handle of the Ch1 waveform). This caused a segmentation fault (see attached video).

https://user-images.githubusercontent.com/108100309/194123422-43d421ec-2776-44f1-9d95-83b7901ae03e.mp4

azonenberg, 2022-10-05:
Is this reproducible?

Please post a backtrace and output of running with --debug up to the point of the failure.

dan-gies, 2022-10-05:
Yes it is. I'll grab that.

bvernoux, 2022-10-05:
A good hint to debug with mingw64 is to do such steps in mingw64 shell
```
cd ~
cd glscopeclient_build
winpty gdb ./glscopeclient.exe
set pagination off
set args --debug
break abort
run
```

When it break you can use following gdb commands to have different backtrace
```
bt
frame
where
thread apply all bt
```

dan-gies, 2022-10-05:
[backtrace](https://github.com/glscopeclient/scopehal/files/9717859/bt.txt)

azonenberg, 2023-09-29:
Closing as glscopeclient is deprecated now. File a new ticket for any crashes in ngscopeclient.

closed 2023-09-29 by azonenberg