ngscopeclient/scopehal #711: Remove usages of non-portable functions/headers

ehntoo, 2022-10-08:
Several places in the codebase make use of POSIX headers and functions (e.g. unistd.h, fcntl.h, and dirent.h). These will make a straight windows port more difficult and should be removed.

An incomplete list:
- [ ] https://github.com/glscopeclient/scopehal/blob/e807bfb/scopehal/Oscilloscope.cpp#L40-L43
- [ ] https://github.com/glscopeclient/scopehal/blob/e807bfb/scopehal/SCPITMCTransport.cpp#L39-L40
- [ ] https://github.com/glscopeclient/scopehal/blob/e807bfb/scopehal/SCPINullTransport.cpp#L39-L40

azonenberg, 2022-10-08:
SCPITMCTransport should be entirely ifdef'd in its current form; it relies on the Linux /dev/usbtmc kernel driver and is not usable on any other platform.

See #519.

azonenberg, 2022-10-08:
SCPINullTransport does not even call any functions from these headers, likely copy-paste bug. So that's trivial to remove.

closed 2022-10-08 by azonenberg