ngscopeclient/scopehal #721: Siglent binary waveforms are unsupported

dan-gies, 2022-11-04:
## System Info:
Glscopeclient version: 0.1-59ef026
OS: Windows 10
Oscilloscope: Siglent SDS6204A (offline, open from .BIN file)

## Description:
I'm unable to open raw waveform binaries using glscopeclient: `ERROR: Unknown vendor in file header`.

Is this planned to be supported? I think this would be useful, since glscopeclient doesn't support >12.5Mpts live captures, and my scope has 500Mpts of memory depth.

dan-gies, 2022-11-04:
Worth noting: Siglent provides a Bin2CSV converter, but it would be nice to import these waveforms natively. However, if this isn't trivial, the converter provides a workaround.

azonenberg, 2022-11-04:
Do you have a link to a spec for the format, and can you share some example files?

Writing an import filter is pretty trivial (the Agilent/Keysight BIN importer is ~200 lines if you don't count the boilerplate for filter creation) but obviously we have to know how the files are structured. If not we might have to reverse engineer it.

dan-gies, 2022-11-04:
The spec can be found here: https://siglentna.com/wp-content/uploads/dlm_uploads/2021/09/How-to-Extract-Data-from-the-Binary-File.pdf. The 6000 series isn't listed as supported by the document, but I'm hoping that the format is unchanged.

I'll need  to grab a smaller binary - max upload is 25 MB.

dan-gies, 2022-11-04:
I split the ch1 capture into 10M volumes using 7zip. If this is unwieldly, let me know.

[ch1-pt1.zip](https://github.com/glscopeclient/scopehal/files/9940048/ch1-pt1.zip)
[ch1-pt2.zip](https://github.com/glscopeclient/scopehal/files/9940054/ch1-pt2.zip)

hansemro, 2023-07-30:
Here is a newer version of the siglent binary spec that specifies SDS2000X HD, SDS5000, and SDS6000: https://www.siglenteu.com/wp-content/uploads/2021/08/How-to-Extract-Data-from-the-Binary-File.pdf

Despite the date in the link, this spec was updated in 2023.

hansemro, 2023-08-01:
I am working on a python script to parse and visualize the waveform: https://gist.github.com/hansemro/4b8733cf5a217bcee7a5796f8eff2726

This script currently supports:
- V2 waveform format with analog + digital waveform data processing and plotting.
    - SDS2000X+ with firmware 1.2.2 to 1.3.9RX
    - SDS6000A  with firmware older than 1.4.1.0 and math disabled
        - tested single channel analog data from @dan-gies
        - a newer V2 format revision includes math waveform support, but that is not yet supported.
    - SDS6000H10/SDS6000H12 should be supported, but may require adjustments to code_per_div.

Unfortunately, I have not yet found a method to identify the model from the bin file. For now, the model argument must be specified to get the correct code_per_div.

[image]

[image]

hansemro, 2023-08-02:
Updated script to support V4 waveform format, which includes support for most scopes running up-to-date firmware.

Fortunately V4 is not much different to V2, so we can greatly simplify the C++ implementation:
- v2_data[0x4:0x400] should be the same as v4_data[0x8:0x404].
- V2 waveform data begins at 0x800, whereas V4 waveform data begins at 0x1000.

Waveform data includes any enabled math channels. Math channel data is inserted between analog and digital channel data.

hansemro, 2023-08-02:
Updated dev branch with support to import math and digital waveforms.

[image]