ngscopeclient/scopehal #723: Protocol Analyzer: doesn't list all USB 2.0 SETUP packets

dan-gies, 2022-11-07:


azonenberg, 2022-11-07:
Moving issue to scopehal as this is likely a back-end issue with the protocol decode class, not the GUI protocol analyzer.

azonenberg, 2022-11-07:
Confirmed it decodes the same for me. The "unknown end PID" is likely something to do with it, but I won't know for sure until I dig deeper (at $dayjob right now so can't spend time on it).

dan-gies, 2022-11-07:
No rush. Note that all 6 `SETUP` packets don't receive an `ACK` (which is the issue I'm diagnosing).

azonenberg, 2022-11-07:
Yes. This might be a case of the decode correctly handling valid traffic but choking on out of spec data.

Generally I try to err on the side of "decode malformed data anyway, but color code or otherwise flag it as something being wrong". So there's probably room to improve.

dan-gies, 2022-11-07:
That seems like a valid approach. The thing that's confusing me here is that the protocol analyzer shows _some_ `SETUP` packets and not others, when they are all identical.

azonenberg, 2022-11-09:
+ @miek since you probably have more low level familiarity with USB than me. Is this traffic malformed or not?

dan-gies, 2022-11-09:
I think it is, since the USB hub is failing to enumerate with the peripheral.

miek, 2022-12-05:
> * @miek since you probably have more low level familiarity with USB than me. Is this traffic malformed or not?

Well.. yes and no. The device should be responding with either an ACK or NAK, but it's also a perfectly valid path through the state machine for the host to timeout waiting for a response and try again.

I've put a proposed fix in #741 that seems to work well:

[image]

PR #741 (2022-12-06): USB2PacketDecoder: fix DecodeSetup behaviour when the ACK/NAK packet is missing
Previously, when looking for a response to a setup packet, DecodeSetup would consume the subsequent packet whether it was an ACK/NACK or not.

In cases where the device did not reply, the decoder would consume a packet from the next transaction and break the decoding.

This commit makes sure that DecodeSetup only consumes the subsequent packet iff it is an ACK or NAK.

fixes #723