ngscopeclient/scopehal #736: Make color a per-stream property rather than per channel

bvernoux, 2022-11-27:


azonenberg, 2022-11-27:
The implementation of this is straightforward, the UI side is more of an open question.

The channel color is used in, for example, the filter graph editor to distinguish filters from each other. We also do want to allow related channels to be visually tied together (e.g. mag and angle of the same S-parameter import).

bvernoux, 2022-11-29:
Having RGB color for each stream seems the best in worst case all color will be identical and that will work for all cases

azonenberg, 2026-02-01:
I'll have a look at the code in a bit, busy at the moment.

In general, scaling is a per-stream property because you can have streams coming out of say an import filter with wildly different scales, or streams with different units (like mag/angle for S-parameters). I'm not opposed to adding some kind of UI mechanism to lock scaling of several streams together so they change in tandem, but it'd take some work to figure out how to do that.