ngscopeclient/scopehal #755: SPIRV linking missing for non-WIN32?

CodingMarco, 2023-03-14:
When I try to build the upstream repo on Linux, I get linker errors like
`/usr/bin/ld: ../../lib/scopehal/libscopehal.so: undefined reference to spvtools::[...]`.

I can see that SPIRV is being linked on WIN32: https://github.com/glscopeclient/scopehal/blob/master/scopehal/CMakeLists.txt#L180
but not on the other platforms.

When I add SPIRV, SPIRV-Tools-opt and SPIRV-Tools to the else() block, building and running glscopeclient works.
Am I missing something or is this a bug?

bvernoux, 2023-03-15:
It seems you have not followed exactly the steps in https://www.antikernel.net/temp/glscopeclient-manual.pdf especially 3.4.1 Linux
(Vulkan SDK install with path) as such binaries are present in Linux version of the SDK and was not present only in Windows version it is why we are rebuilding all for Windows...

azonenberg, 2023-03-21:
Can we confirm this is fixed?

CodingMarco, 2023-03-21:
Yes, thanks for the hint to the manual.

closed 2023-03-21 by CodingMarco