ngscopeclient/scopehal #784: [Siglent SDS2000X Plus] Partially broken 8/10-bit mode switching; Slow data acquisition in 10-bit mode

hansemro, 2023-07-24:
## Setup

- Arch Linux
- scopehal-apps (master: [`18a4876430`](https://github.com/glscopeclient/scopehal-apps/commit/18a487643006002bcc1e927dcdf22cd51711f774))
- scopehal (master: 148899e23459d7417414a6cc3727053ade69ae77)
- SDS2104X Plus (with no software upgrades) connected via lan
    - Tested versions:
        - v1.3.9R10
        - v1.3.9R12
        - v1.5.2R3

## Issue Description

- The `:WAVEFORM:WIDTH` setting does not update between `BYTE` and `WORD` when changing between 8- and 10-bit ADC modes, respectively.
    - Get/SetADCMode should have sent the command to set width, but the setting does not change.
        - Need to investigate why it fails and where it should be set.
        - Use `:WAVEFORM:WIDTH?` in SCPI console to query the width after changing the setting.
    - This results in invalid waveform rendering and timescale when WIDTH is not correctly assigned.
    - Example invalid 8-bit waveform with `WORD` width: [image]
    - Expected 8-bit waveform with `BYTE` width: 
[image]
    - Can be fixed temporarily by manually setting waveform width in SCPI console after changing ADC mode.

- [minor issue] Get/SetADCMode should not expect a reply when sending a set command.
    - Fixing this eliminates some `Socket read failed` warnings and improves minor slow downs when switching between modes.
    - https://github.com/glscopeclient/scopehal/blob/148899e23459d7417414a6cc3727053ade69ae77/scopehal/SiglentSCPIOscilloscope.cpp#L3245
    - https://github.com/glscopeclient/scopehal/blob/148899e23459d7417414a6cc3727053ade69ae77/scopehal/SiglentSCPIOscilloscope.cpp#L3251
    - https://github.com/glscopeclient/scopehal/blob/148899e23459d7417414a6cc3727053ade69ae77/scopehal/SiglentSCPIOscilloscope.cpp#L3281
    - https://github.com/glscopeclient/scopehal/blob/148899e23459d7417414a6cc3727053ade69ae77/scopehal/SiglentSCPIOscilloscope.cpp#L3286

- Slow data acquisition while in 10-bit mode due to expecting twice more data than received from the scope.
    - Data acquisition and processing (for single trigger) takes ~15.5 sec per enabled channel!
    - Issue related to `hdSizeWorkaround` in [SiglentSCPIOscilloscope::ReadWaveformBlock](https://github.com/glscopeclient/scopehal/blob/148899e23459d7417414a6cc3727053ade69ae77/scopehal/SiglentSCPIOscilloscope.cpp#L1448-L1454).
        - With the versions tested and tracing enabled, I see that the waveform data length doubles when switching from `BYTE` to `WORD` waveform width. I suspect the broken SetADCMode lead to the current size workaround.
    - Data acquisition in 10-bit mode always results in `Socket read failed` warnings due to increased data length.
    - By not doubling length of data, we can reduce data acquisition and processing to 228.6 ms per enabled channel!

## Relevant Development Branches

TODO

azonenberg, 2023-07-24:
The current hdSizeWorkaround (tested on 1.3.9R6) is caused by the waveform data length *not* doubling in size on older firmware revisions. Presumably Siglent fixed the issue (we reported it to them).

So we'll need to add a conditional to the code to enable it only on versions older than XX (may need to try a few versions and/or read release notes)

hansemro, 2023-07-24:
Unfortunately, since I have a newer hardware revision (05-05) and cannot downgrade to versions below 1.3.9R10 (due to rollback protection), I cannot confirm the bug in older versions. Release notes do not clearly specify fixing this bug, but it seems fixed from what I can tell.

For now, I will just assume versions 1.3.9R6 and older will require the workaround.

hansemro, 2023-07-24:
Updated post with [python script](https://gist.github.com/hansemro/e5d3fd1bca6d77f0093db05071f69991) to reproduce :WAV:WIDTH bug.

azonenberg, 2023-07-24:
+ @mubes 

Can you test this on your older hardware rev 2000X+?

mubes, 2023-07-24:
Yes, I'm on vacation the end of this week so should have some time.

PR #785 (2023-10-02): [Siglent SDS2000X Plus] Fix ADC mode setting; Fix slow 10-bit data acquisition with newer firmware
Fixes #784.

Tasks:
- [x] Add delays to set :WAV:WIDTH
- [x] Replace invalid queries on set commands
- [x] Skip data size workaround for firmware versions 1.3.9R10 and newer. 
    - I currently assume data size bug is present in versions 1.3.9R6 and older.
- [ ] Get confirmation that the size bug is or is not present in versions older than 1.3.9R6.
- [ ] Test from another SDS2000X Plus user.
closed 2023-10-02 by azonenberg