ngscopeclient/scopehal #815: Stack smashing or FPE when changing coupling DC > Ground > DC

juh2600, 2023-11-20:
Observed in Linux Release build of ngscopeclient ecc525e, scopehal 2f49730, using a Rigol DS1054Z connected via LAN.

To reproduce:

1. Clear preferences and stuff: `rm -rf ~/.config/ngscopeclient/`
2. Hook up Rigol DS1054Z chan1 to signal source and press AUTO to reset scope config
3. Launch `ngscopeclient --debug --trace SCPISocketTransport`
4. Add > Oscilloscope > Connect > rigol, lan
5. Trigger a one-shot capture from ngscopeclient
6. Double-click yellow CHAN1 marker in top left of waveform group
7. Change Input > Coupling from DC to Ground
8. Trigger another one-shot capture from ngscopeclient [bonus: sometimes this produces a SIGFPE, sometimes it doesn't!]
9. Change Input > Coupling from Ground back to DC
10. Trigger another one-shot capture from ngscopeclient

Final words on the console:

```
[SCPISocketTransport::SendCommand]     [10.8.0.207] Sending WAV:STAR 41
[SCPISocketTransport::SendCommand]     [10.8.0.207] Sending WAV:STOP 1200
[SCPISocketTransport::SendCommand]     [10.8.0.207] Sending WAV:DATA?
[SCPISocketTransport::ReadRaw*** stack smashing detected ***: terminated
fish: Process 344989, 'ngscopeclient' from job 1, 'ngscopeclient --debug --trace S…' terminated by signal SIGABRT (Abort)
Done in 52s 970ms
~/scope/ rm -rf ~/.config/ngscopeclient/; ngscopeclient --debug --trace SCPISocketTransport | tee coupling-change-stack-smashing
```

I expect we'll need more detailed debug information to diagnose this, but I don't know what info to grab or how to grab it.

[coupling-change-stack-smashing.txt](https://github.com/ngscopeclient/scopehal/files/13405572/coupling-change-stack-smashing.txt)
[couple-gnd-sigfpe.txt](https://github.com/ngscopeclient/scopehal/files/13405608/couple-gnd-sigfpe.txt)

juh2600, 2023-11-20:
Going DC -> AC -> DC works as expected, but DC -> AC -> DC -> GND -> AC crashes as above, so it seems moving away from GND is important here.

juh2600, 2023-11-20:
[stack-smashing-filtered.pcap.gz](https://github.com/ngscopeclient/scopehal/files/13406119/stack-smashing-filtered.pcap.gz)
PCAP showing stack smashing traffic

azonenberg, 2023-11-20:
Can you file a separate ticket for the SIGFPE issue and link the FPE log there? I'm starting to think these are two separate bugs so let's track as such.

juh2600, 2023-11-20:
As of 3a5636c3c56513ea0055add252aa79f059bc04c2, the procedure in the first comment now behaves as desired (no app crashes).

closed 2023-11-20 by juh2600