ngscopeclient/scopehal #838: Tektronix MSO64 Connect Error

codepox, 2024-01-02:
MSO64 Software 2.4.4.1152
Connect over Ethernet using transport lxi
I get the following error and ngscopeclient just close

Warning: Malformed color string ""
Error: Read error (timeout)
terminate called after throwing an instance of 'std::invalid_argument'
  what():  stof
Aborted

codepox, 2024-01-02:
I also get these warnings and errors when acquiring waveform

Warning: Malformed color string ""
    Warning: Preamble error (read only 12 fields from 24 semicolons)
Warning: Malformed color string ""
Warning: Malformed color string ""
    Warning: Preamble error (read only 12 fields from 24 semicolons)
    ERROR: Retried too many times acquiring channel
Warning: Malformed color string ""

azonenberg, 2024-01-02:
Can you try running with `--debug --trace SCPILxiTransport` and upload the log?

Also, try using the "lan" transport rather than LXI (you may need to change settings on the scope to enable the socket server, and set the protocol as "none" rather than "terminal" which is the default).

codepox, 2024-01-03:
Here is the trace with lxi transport

OMP_WAIT_POLICY not set to PASSIVE
Re-exec'ing with correct environment
Initializing Vulkan
    VK_KHR_get_physical_device_properties2: supported
    VK_KHR_xcb_surface: supported
    VK_KHR_xlib_surface: supported
    VK_EXT_debug_utils: supported
    Loader/API support available for Vulkan 1.3
    Vulkan 1.2 support available, requesting it
    Initializing glfw 3.3.6 X11 GLX EGL OSMesa clock_gettime evdev shared
    GLFW required extensions:
        VK_KHR_surface
        VK_KHR_xcb_surface
    Physical devices:
        Device 0: AMD Radeon RX 6600 (RADV NAVI23)
            API version:            0x004030ee (0.1.3.238)
            Driver version:         0x05c00004 (0.23.0.4)
            Vendor ID:              1002
            Device ID:              73ff
            Device type:            Discrete GPU
            int64:                  yes
            int16:                  yes (allowed in SSBOs)
            int8:                   yes (allowed in SSBOs)
            Max image dim 2D:       16384
            Max storage buf range:  4095 MB
            Max mem alloc:          4095 MB
            Max compute shared mem: 64 KB
            Max compute grp count:  65535 x 65535 x 65535
            Max compute invocs:     1024
            Max compute grp size:   1024 x 1024 x 1024
            Memory types:
                Type 0
                    Heap index: 0
                    Device local
                Type 1
                    Heap index: 0
                    Device local
                Type 2
                    Heap index: 1
                    Host visible
                    Host coherent
                Type 3
                    Heap index: 2
                    Device local
                    Host visible
                    Host coherent
                Type 4
                    Heap index: 2
                    Device local
                    Host visible
                    Host coherent
                Type 5
                    Heap index: 1
                    Host visible
                    Host coherent
                    Host cached
                Type 6
                    Heap index: 1
                    Host visible
                    Host coherent
                    Host cached
                Type 7
                    Heap index: 0
                    Device local
                    Device coherent
                    Device uncached
                Type 8
                    Heap index: 1
[... 399 lines cut ...]
[SCPILxiTransport::SendCommand] Sending HOR:MODE:SAMPLER?
[SCPILxiTransport::ReadReply] Got 1.2500E+9
Warning: Malformed color string ""
[SCPILxiTransport::SendCommand] Sending CH1:OFFS?
[SCPILxiTransport::ReadReply] Got 0.0E+0
[SCPILxiTransport::SendCommand] Sending CH1:SCA?
[SCPILxiTransport::ReadReply] Got 50.0000E-3
[SCPILxiTransport::SendCommand] Sending CH2:OFFS?
[SCPILxiTransport::ReadReply] Got 0.0E+0
[SCPILxiTransport::SendCommand] Sending CH2:SCA?
[SCPILxiTransport::ReadReply] Got 50.0000E-3
[SCPILxiTransport::SendCommand] Sending CH3:OFFS?
[SCPILxiTransport::ReadReply] Got 0.0E+0
[SCPILxiTransport::SendCommand] Sending CH3:SCA?
[SCPILxiTransport::ReadReply] Got 50.0000E-3
[SCPILxiTransport::SendCommand] Sending CH4:OFFS?
[SCPILxiTransport::ReadReply] Got 0.0E+0
[SCPILxiTransport::SendCommand] Sending CH4:SCA?
[SCPILxiTransport::ReadReply] Got 50.0000E-3
[SCPILxiTransport::SendCommand] Sending DISP:SPECV:CH1:VERT:POS?
Error: Read error (timeout)
[SCPILxiTransport::ReadReply] Got  
terminate called after throwing an instance of 'std::invalid_argument'
  what():  stof
Aborted

codepox, 2024-01-03:
Here is the trace with lan

OMP_WAIT_POLICY not set to PASSIVE
Re-exec'ing with correct environment
Initializing Vulkan
    VK_KHR_get_physical_device_properties2: supported
    VK_KHR_xcb_surface: supported
    VK_KHR_xlib_surface: supported
    VK_EXT_debug_utils: supported
    Loader/API support available for Vulkan 1.3
    Vulkan 1.2 support available, requesting it
    Initializing glfw 3.3.6 X11 GLX EGL OSMesa clock_gettime evdev shared
    GLFW required extensions:
        VK_KHR_surface
        VK_KHR_xcb_surface
    Physical devices:
        Device 0: AMD Radeon RX 6600 (RADV NAVI23)
            API version:            0x004030ee (0.1.3.238)
            Driver version:         0x05c00004 (0.23.0.4)
            Vendor ID:              1002
            Device ID:              73ff
            Device type:            Discrete GPU
            int64:                  yes
            int16:                  yes (allowed in SSBOs)
            int8:                   yes (allowed in SSBOs)
            Max image dim 2D:       16384
            Max storage buf range:  4095 MB
            Max mem alloc:          4095 MB
            Max compute shared mem: 64 KB
            Max compute grp count:  65535 x 65535 x 65535
            Max compute invocs:     1024
            Max compute grp size:   1024 x 1024 x 1024
            Memory types:
                Type 0
                    Heap index: 0
                    Device local
                Type 1
                    Heap index: 0
                    Device local
                Type 2
                    Heap index: 1
                    Host visible
                    Host coherent
                Type 3
                    Heap index: 2
                    Device local
                    Host visible
                    Host coherent
                Type 4
                    Heap index: 2
                    Device local
                    Host visible
                    Host coherent
                Type 5
                    Heap index: 1
                    Host visible
                    Host coherent
                    Host cached
                Type 6
                    Heap index: 1
                    Host visible
                    Host coherent
                    Host cached
                Type 7
                    Heap index: 0
                    Device local
                    Device coherent
                    Device uncached
                Type 8
                    Heap index: 1
[... 172 lines cut ...]
[SCPITransport::SendCommandQueued] 32 commands now queued
[SCPITransport::SendCommandQueued] 33 commands now queued
[SCPITransport::SendCommandQueued] 34 commands now queued
[SCPITransport::SendCommandQueued] 35 commands now queued
[SCPITransport::SendCommandQueued] 36 commands now queued
[SCPITransport::SendCommandQueued] 37 commands now queued
[SCPITransport::SendCommandQueued] 38 commands now queued
[SCPITransport::SendCommandQueued] 39 commands now queued
[SCPITransport::SendCommandQueued] 40 commands now queued
[SCPITransport::SendCommandQueued] 41 commands now queued
[SCPITransport::SendCommandQueued]     42 commands now queued
[SCPITransport::SendCommandQueued]     43 commands now queued
[SCPITransport::SendCommandQueued]     44 commands now queued
[SCPITransport::SendCommandQueued]     45 commands now queued
[SCPITransport::SendCommandQueued]     46 commands now queued
[SCPITransport::FlushCommandQueue] 46 commands being flushed
QueueManager creating family=1 index=2 name=WaveformThread.queue
[SCPITransport::SendCommandQueued] 1 commands now queued
[SCPITransport::SendCommandQueued] 2 commands now queued
[SCPITransport::FlushCommandQueue] 2 commands being flushed
Warning: Malformed color string ""
Warning: Socket read failed (errno=11, Resource temporarily unavailable)
terminate called after throwing an instance of 'std::invalid_argument'
  what():  stof
Aborted