ngscopeclient/scopehal #858: BK Precision 9103 power supply

rgov, 2024-02-13:
This power supply supports a serial protocol described in the [Programming Manual](https://bkpmedia.s3.us-west-1.amazonaws.com/downloads/programming_manuals/en-us/9103_9104_programming_manual.pdf).

Here are examples of the basic commands. Note that volts and amps values are multiplied by 100.

| Command | Example output | Explanation |
| --- | --- | -- | 
| `SOUT 1` |  | Set output on |
| `SOUT 0` |  | Set output off |
| `GOUT` | `0` | Output is `0`=off, `1`=on |
| `GETD` | `1002` `0030` `0` | Presently outputting 10.02V, 0.30A, `0`=CV mode vs `1`=CC mode |
| `GETS 3` | `1000` `0045` | Present settings (whether outputting or not) are 10.00V, 0.45A |
| `VOLT` | `3` `1000` | Set present voltage setting to 10.00V (note `3` prefix) |
| `CURR` | `3` `0021` | Set present current setting to 0.21A |
| `GOVP` | `4270` | Upper limit of voltage is 42.70V |
| `GOCP` | `2040` | Upper limit of current is 20.40A |

These can be tested at the command line with:

```
$ picocom -b 9600 --echo --omap lfcr --imap crcrlf /dev/cu.usbserial-0001
```

---

Above I've ignored the "preset" feature of the power supply, which is complicated, but in brief: Many of the command operands take a prefix of 0-3, where 3 means "the value it is set to now" and 0-2 mean "the stored value for preset A, B, or C". For example: `VOLT 01000` sets Preset A voltage to 10.00V, whereas `VOLT 31000` sets the output voltage to 10.00V.

Presets can also be used to create a program that switches between presets automatically at intervals.

rgov, 2024-02-13:
I think there's an undocumented command `GMOD` which returns some kind of model identifier. Their PSCS program has some logic that tries to identify the model using the first digit. Pseudocode:

```java
      if (response.startsWith("8"))
        return "SSP" + response; 
      if (response.startsWith("6"))
        return "KPS" + response; 
      if (response.startsWith("NTP5"))
        return "NTP" + response.substring(3); 
      if (response.startsWith("SDP"))
        return "SDP" + response.substring(3); 
      return response.startsWith("-", "");
```

My model returns `8320` which would translate to the model `SSP8320` which points to (*drumroll*) the [Manson SSP-8320](https://www.manson.com.hk/product/ssp-8320/). Gasp!

[This Python project](https://github.com/tsitle/manson_ps_serial_lib) (N.B.: GPLv3) has more info about all the Manson models and which ones are resold as BK Precision.

azonenberg, 2024-02-13:
Doesn't look too complicated, I can try to throw something together quickly when I get a chance. But it's always tricky to develop without having hands on the hardware.