ngscopeclient/scopehal #867: MoltenVK throws an exception when built for macOS on x86

sslupsky, 2024-04-12:
It built the app with no build errors.
macOS Sonoma 14.4.1
MP 6,1

When I run the app, it throws an error. Not sure if this is a problem with my Homebrew installation of Vulkan or if there is some other issue. There was no app bundle, just a "ngscopeclient" executable. Is that the normal build output for macOS? Or, is there another step to build the app bundle that I missed?

```
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[BronzeMtlRenderCmdEncoder setVertexBuffer:offset:attributeStride:atIndex:]: unrecognized selector sent to instance 0x7fcc1f8abc00'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007ff81607d2ee __exceptionPreprocess + 242
	1   libobjc.A.dylib                     0x00007ff815b6e21b objc_exception_throw + 48
	2   CoreFoundation                      0x00007ff81612265d -[NSObject(NSObject) __retain_OA] + 0
	3   libMoltenVK.dylib                   0x0000000181252f75 _ZNSt3__110__function6__funcIZN39MVKGraphicsResourcesCommandEncoderState10encodeImplEjE3$_6NS_9allocatorIS3_EEFvP17MVKCommandEncoderR19MVKMTLBufferBindingEEclEOS7_S9_ + 69
	4   libMoltenVK.dylib                   0x000000018124f704 _ZN39MVKGraphicsResourcesCommandEncoderState14encodeBindingsE14MVKShaderStagePKcbNSt3__18functionIFvP17MVKCommandEncoderR19MVKMTLBufferBindingEEENS4_IFvS6_S8_11MVKArrayRefIKjEEEENS4_IFvS6_R20MVKM	5   libMoltenVK.dylib                   0x0000000181250110 _ZN39MVKGraphicsResourcesCommandEncoderState10encodeImplEj + 272
	6   libMoltenVK.dylib                   0x00000001811bd4ec _ZN17MVKCommandEncoder17finalizeDrawStateE16MVKGraphicsStage + 76
	7   libMoltenVK.dylib                   0x00000001811b8193 _ZN17MVKCmdDrawIndexed6encodeEP17MVKCommandEncoder + 595
	8   libMoltenVK.dylib                   0x00000001811bb0ca _ZN17MVKCommandEncoder6encodeEPU27objcproto16MTLCommandBuffer11objc_objectP25MVKCommandEncodingContext + 250
	9   libMoltenVK.dylib                   0x00000001811bb52c _ZN16MVKCommandBuffer6submitEP31MVKQueueCommandBufferSubmissionP25MVKCommandEncodingContext + 236
	10  libMoltenVK.dylib                   0x0000000181229676 _ZN35MVKQueueFullCommandBufferSubmissionILm1EE20submitCommandBuffersEv + 102
	11  libMoltenVK.dylib                   0x0000000181227d76 _ZN31MVKQueueCommandBufferSubmission7executeEv + 198
	12  libMoltenVK.dylib                   0x00000001812261a9 _ZN8MVKQueue6submitI12VkSubmitInfoEE8VkResultjPKT_P9VkFence_T13MVKCommandUse + 329
	13  libMoltenVK.dylib                   0x0000000181171e7a vkQueueSubmit + 74
	14  ngscopeclient                       0x000000010e183b46 _ZN12VulkanWindow6RenderEv + 2742
	15  ngscopeclient                       0x000000010e1cf842 main + 850
	16  dyld                                0x00007ff815ba2366 start + 1942
)
libc++abi: terminating due to uncaught exception of type NSException
zsh: abort      ./ngscopeclient
```

d235j, 2024-04-12:
Are you using OpenCore Legacy Patcher to run an unsupported macOS version? If so, this is a known issue that I have reproduced and is due to bugs in the patches. I have reported it upstream to OCLP via their Discord, but so far there has been no fix.

sslupsky, 2024-04-12:
Ahh, indeed I am.  I started to wonder about that.  Thanks for the feedback.  I'll close since this is an "unsupported" configuration and I shouldn't expect anyone to do anything about this.

closed 2024-04-12 by sslupsky

d235j, 2024-04-12:
> There was no app bundle, just a "ngscopeclient" executable. Is that the normal build output for macOS? Or, is there another step to build the app bundle that I missed?

We do not have an app bundle yet — this is normal.