ngscopeclient/scopehal #881: Changing units causes histogram to either not display data or to crash

mx-shift, 2024-09-02:
Built at 58af58e on Fedora 39.  nVidia Quadro P600 with nVidia 545.29.06 drivers.

If I import [00.0.revolution1.500_flashfloppy_v341.csv](https://github.com/ngscopeclient/scopehal-apps/files/14317939/00.0.revolution1.500_flashfloppy_v341.csv), I get a waveform display with the Y units default to volts.  I can create a histogram and get data that appears to be correct.

The Y units are actually phase error measured in clock ticks so I tried to guess a more appropriate unit.  Choosing either UI or dimensionless before creating a histogram, the waveform renders fine but the histogram is blank:
[image]

If I change the Y units to UI or dimensionless after opening a histogram, I get a crash:
```
mxshift@eps-conduit .../build/src/ngscopeclient ❯ ./ngscopeclient
Floating point exception (core dumped)
```

azonenberg, 2024-09-02:
Moved to scopehal as this is an issue (or series of issues) with the histogram filter.

azonenberg, 2024-09-02:
The reason nothing is displayed is that it's trying to create a histogram with only one bin (and apparently rendering breaks in that edge case).

```
[HistogramFilter::Refresh] nmin = -24 UI, nmax = 2 UI
[HistogramFilter::Refresh] Calculated: 1 bins
[HistogramFilter::Refresh] Bin size: 28.6 UI
```

azonenberg, 2024-09-02:
And it turns out the reason for this is that autoranging does not actually automatically calculate the bin size. You're asking for 100-UI wide bins by default so it's getting very confused.

Fix pending, will need to automatically choose a sane bin size if in autorange mode.

commit ae8f81c6 (2024-09-02): HistogramFilter: improve calculation of bin sizes when autoranging, sanity check to prevent impossibly-small bin sizes etc. Fixes #881.
closed 2024-09-02 by azonenberg