ngscopeclient/scopehal #887: narrowing conversion error on aarch64 build

erdnaxe, 2024-09-11:
Compilation on aarch64 linux machines fails with multiple errors like this one:
```
/build/source/lib/scopehal/base64.cpp:12:44: error: narrowing conversion of '-1' from 'int' to 'char' [8;;https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wnarrowing-Wnarrowing8;;]
   12 |         static const char decoding[] = {62,-1,-1,-1,63,52,53,54,55,56,57,58,59,60,61,-1,-1,-1,-2,-1,-1,-1,0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,-1,-1,-1,-1,-1,-1,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51};
      |                                            ^~
```

According to https://stackoverflow.com/questions/31634962/narrowing-conversion-from-int-to-char-inside-for-legal-values-when-cro, https://github.com/ngscopeclient/scopehal/blob/82c4369a1d976030675caf258b89f56979114f33/scopehal/base64.cpp#L12 is missing a `signed` keyword to be robust when targeting aarch64.

azonenberg, 2024-09-11:
Oops. Interesting that this doesn't seem to fail on MacOS arm64 (apple silicon) machines.

Send a PR with the fix?

azonenberg, 2024-09-11:
Hopefully there's not too many more similar bugs lurking. I generally used uint8_t rather than char for single-byte variables, and char only for textual data.

closed 2024-09-12 by azonenberg