ngscopeclient/scopehal #943: Question: Rigol oscilloscope driver refactoring

diggit, 2024-11-23:
Hi,
I have some WIP changes in Rigol oscilloscope driver. It implements sampling rate/memory depth settings, faster data readout (tested on DS1000Z and more) and possibly more changes. I ended up with the need to with distinguishing between each scope series, not just protocol versions (MS05, DS, DS_OLD,...) due to some properties being only for specific series. It is quite heavy refactor of the existing code.

I did these changes longer time ago, and they need porting to current code-base to be "upstreamable". I can test these changes on DS1000Z and possibly on MSO/DS2000A.

My question is, if such heavy refactoring, which could temporarily break support of some Rigol scopes, would be even considered for merging.

azonenberg, 2024-11-23:
Yes, it would be considered although we'd want to get as many Rigol users as possible to test and fix issues with other scope families (e.g. MSO5000) before merging.

You have time, we're doing feature freeze for v0.1 end of month so we won't be taking new major PRs until after v0.1 releases unless they're bug fixes, documentation, or OS packaging related.

OllyFunkster, 2025-04-16:
Did anything become of this? I've found that the current driver doesn't work at all with my MSO2302A and although I've managed to make some hacky tweaks to get it to work, I wonder if the refactored branch is available anywhere so I could provide some test results against it?

diggit, 2025-04-17:
@OllyFunkster sorry no progress on my side. My code is based on very old version and not published.
Anyway my changes did not take MSO2000 into account as I don't own one.

Key change in my code was to differentiate between every model series and then group some part together `switch` came in very handy. Current Protocol differentiation is not sufficient. Then I implemented sampling rate switching in somewhat hacky way.

At this moment, I lack motivation to work on this (no projects needing extensive oscilloscope usage). Feel free to open PR with your changes. Then they can be discussed.