ngscopeclient/scopehal #948: Add binomial filtering mode to moving average filter

MatFi, 2024-12-22:
The current moving average filter implementation could be enhanced by adding support for binomial filtering as an additional mode. Binomial filtering is preserving signal features compared to simple moving averages.

#### Benefits of binomial filtering:
* Better preservation of signal shape compared to simple moving average
* No phase shift (symmetric filter)

#### Technical Considerations

* Need to handle edge cases at the start/end of the waveform
* Consider optimal buffer sizes for variable window lengths
* Computationally efficient implementation possible using Pascal's triangle, coefficients causing just a few extra multiplications I guess
* When implemented properly we will get a [kernel-smoother](https://en.wikipedia.org/wiki/Kernel_smoother) framework. 

#### Kontext

This year, I came into possession of a Pico 3000 that offers a quit good sampling rate of up to 1 GS/s. However, I find that the limitation to 8 vertical bits does present a challenge. It might be beneficial to consider implementing  proper averaging, which has the potential to enhance voltage resolution, albeit at the expense of temporal resolution. Given the scope's relatively modest bandwidth of 70 MHz, I believe this will be a reasonable trade-off in such cases.

----------------------
Hey, I'd love to send through a pull request! But before I do, it'd be great if you could let me know if you're interested in such a contribution, or if there are any extra things/hints/warnings I should consider before I get going.

MatFi, 2024-12-25:
After further consideration and a first look into the exiting codebase, I would like to propose a more flexible approach:

Instead of implementing specific filter types, we could create a generic `KernelFilter` class that:
1. Offers a selection of predefined kernels
2. Allows users to define custom kernels
3. Uses efficient FFT-based implementation under the hood

Benefits of this approach:
- More versatile: Can handle multiple filter types with a single implementation
- More efficient: Combines multiple operations in one pass
- More extensible: Easy to add new filter types

Use cases would include:
1. Signal smoothing
2. Numerical differentiation (as proposed in https://github.com/ngscopeclient/scopehal/issues/140
3. Combined operations (e.g., Savitzky-Golay filtering, which performs smoothing and differentiation simultaneously)
4. General FIR filtering (e.g., low-pass, high-pass, band-pass filters) by defining appropriate kernel coefficients ( Create FFT convolution filter for applying narrowband filters etc #624 ) 
This would provide a more maintainable and future-proof solution while potentially improving performance through FFT-based processing.

azonenberg, 2024-12-25:
Yeah I think ultimately we're going to want two backends, one tap FIR based and one FFT based, and choose which to use based on the kernel size and some heuristics (i.e. is O(M*N) greater than O(N log N), times some implementation-specific scaling factor, for kernel size M).

One of the challenges with #140 is the overhaul of the units framework needed to properly support second-order units like "volts per second". Hence why it's been tabled.

azonenberg, 2024-12-25:
Anyway, feel free to throw more ideas in the ring. I don't have the time to think about this in too much detail as I'm busy getting ready for the v0.1 release but it can definitely be something we work on perhaps on the way to v0.2.