ngscopeclient/scopehal #951: MIL-STD-1553 Decode Not Producing Symbols Or Data

DrAnonymous, 2025-01-20:
Using a known good 1553 source and sampling it with an oscope then feeding to the 1553 decoder does not produce anything. Attached is sample data showing the same command word and 2 data words repeated. This sample data does not have a response status message, but other sample data can be created if needed.

azonenberg, 2025-01-20:
Are you able to share a scopesession and/or CSV export of the offending packet so I can troubleshoot?

DrAnonymous, 2025-01-20:
Apologies I'm having an issue attaching the scopesession.

azonenberg, 2025-01-20:
Just attach a zip of the scopesession plus the _data directory (the data dir is important as that's where the waveforms live, the session is just configuration)

DrAnonymous, 2025-01-20:
Here's the csv export. Where is the _data directory on a windows machine going to be and I'll grab it.

[1553CmdAnd2Data.zip](https://github.com/user-attachments/files/18472026/1553CmdAnd2Data.zip)

azonenberg, 2025-01-20:
It's in the same directory as the session. You should see foo.scopesession and foo_data

DrAnonymous, 2025-01-20:
Sorry. I wasn't paying attention. I have it zipped and will post it shortly. I have to move the files across different computers to get it, since the hardware is on another computer.