ngscopeclient/scopehal #973: Rigol DS1054Z `ERROR: RigolOscilloscope::GetInterleaveConflicts not implemented for this model`

SimonMerrett, 2025-06-13:
First time using ngscopeclient. Thank you very much everyone who has worked on this!

This feels very similar to https://github.com/ngscopeclient/scopehal-apps/issues/790 

I am getting this error constantly streaming in the terminal. I tried `--debug --trace SCPISocketTransport` and this is the kind of print I get. I'm using LAN for transport, although I had similar results with LXI. Built on Ubuntu 24.04.2 LTS in a VirtualBox VM, hosted on Win10. I can get the waveform to appear but It feels like I should be able to do things with the timebase and see / align the waveform with the trigger but the trace is zoomed well out of the waveform and can't retrieve the timebase from the scope. 

```
 ./ngscopeclient --debug --trace SCPISocketTransport
OMP_WAIT_POLICY not set to PASSIVE
Re-exec'ing with correct environment
Initializing Vulkan
    VK_KHR_get_physical_device_properties2: supported
    VK_KHR_xcb_surface: supported
    VK_KHR_xlib_surface: supported
    VK_EXT_debug_utils: supported
    Loader/API support available for Vulkan 1.3
    Vulkan 1.2 support available, requesting it
    Initializing glfw 3.3.10 X11 GLX EGL OSMesa clock_gettime evdev shared
    GLFW required extensions:
        VK_KHR_surface
        VK_KHR_xcb_surface
    Physical devices:
        Device 0: llvmpipe (LLVM 19.1.1, 128 bits)
            API version:            0x00403121 (0.1.3.289)
            Driver version:         0x00000001 (0.0.0.1)
            Vendor ID:              10005
            Device ID:              0000
            Device type:            CPU
            int64:                  yes
            int16:                  yes (allowed in SSBOs)
            int8:                   yes (allowed in SSBOs)
            Max image dim 2D:       16384
            Max storage buf range:  128 MB
            Max mem objects:        4294967295
            Max compute shared mem: 32 KB
            Max compute grp count:  65535 x 65535 x 65535
            Max compute invocs:     1024
            Max compute grp size:   1024 x 1024 x 1024
            Memory types:
                Type 0
                    Heap index: 0
                    Device local
                    Host visible
                    Host coherent
                    Host cached
            Memory heaps:
                Heap 0
                    Size: 20 GB
                    Device local
        Selected device 0
            Queue families (1 total)
                Queue type 0
                    Queue count:          1
                    Timestamp valid bits: 64
                    Graphics
                    Compute
                    Transfer
                    Sparse binding
            Driver: vk::DriverId::eMesaLlvmpipe
            Enabling 64-bit float support
            Enabling 64-bit integer support
            Enabling 16-bit integer support
            Enabling 16-bit integer support for SSBOs
            Enabling 8-bit integer support
            Enabling 8-bit integer support for SSBOs
            Device has VK_KHR_push_descriptor, requesting it
            Device has VK_KHR_shader_non_semantic_info, requesting it
            Device has VK_EXT_memory_budget, requesting it
            Device has VK_EXT_shader_atomic_float, requesting it
            Using heap 0, type 0 for pinned host memory
[... 230 lines cut ...]
[SCPISocketTransport::SendCommand] [192.168.1.175] Sending :TRIG:STAT?
[SCPISocketTransport::ReadReply] [192.168.1.175] Got STOP
ERROR: RigolOscilloscope::GetInterleaveConflicts not implemented for this model
ERROR: RigolOscilloscope::GetInterleaveConflicts not implemented for this model
ERROR: RigolOscilloscope::GetInterleaveConflicts not implemented for this model
ERROR: RigolOscilloscope::GetInterleaveConflicts not implemented for this model
ERROR: RigolOscilloscope::GetInterleaveConflicts not implemented for this model
ERROR: RigolOscilloscope::GetInterleaveConflicts not implemented for this model
ERROR: RigolOscilloscope::GetInterleaveConflicts not implemented for this model
ERROR: RigolOscilloscope::GetInterleaveConflicts not implemented for this model
ERROR: RigolOscilloscope::GetInterleaveConflicts not implemented for this model
ERROR: RigolOscilloscope::GetInterleaveConflicts not implemented for this model
ERROR: RigolOscilloscope::GetInterleaveConflicts not implemented for this model
ERROR: RigolOscilloscope::GetInterleaveConflicts not implemented for this model
ERROR: RigolOscilloscope::GetInterleaveConflicts not implemented for this model
ERROR: RigolOscilloscope::GetInterleaveConflicts not implemented for this model
ERROR: RigolOscilloscope::GetInterleaveConflicts not implemented for this model
ERROR: RigolOscilloscope::GetInterleaveConflicts not implemented for this model
ERROR: RigolOscilloscope::GetInterleaveConflicts not implemented for this model
ERROR: RigolOscilloscope::GetInterleaveConflicts not implemented for this model
ERROR: RigolOscilloscope::GetInterleaveConflicts not implemented for this model
ERROR: RigolOscilloscope::GetInterleaveConflicts not implemented for this model
ERROR: RigolOscilloscope::GetInterleaveConflicts not implemented for this model
ERROR: RigolOscilloscope::GetInterleaveConflicts not implemented for this model
```

azonenberg, 2025-06-13:
Keep it open. As a project with a small team and a huge scope (pun intended), we often have issues sit around for months or even years before someone with the right combination of hardware and experience has the time and motivation to pick them up. But that doesn't make them not real problems/features we want to work on.

azonenberg, 2025-06-13:
And as the project grows, we want new contributors to be able to find things they can work on with their skills and hardware. Keeping all the little nuisances open gives them a good way to get started.