ngscopeclient/scopehal #975: BIN Import Access Violation

I-AM-EXFOLIATING, 2025-06-19:
I've generated a .bin waveform recording using a Rigol MSO5074 containing a 20ms span of the scope's calibration signal on channel 1 and an open input on channel 2, with 1k memory depth at 50kSa/s. Upon import of this file using the BIN Import filter graph element, the ngscopeclient (ran with `--debug` flag) crashes with a segmentation violation. Running on Windows 11.

Terminal output
```
Vendor:    Rigol
Waveforms: 1

Waveform 1:
    Samples:      1000
    Buffers:      1
    Type:         1
    Duration:     0.00 us
    Start:        0.00 us
    Interval:     0.00 ns
    Origin:       0.00 us
    Holdoff:      0.00 ms
    Sample Rate:  1000000.00 Msps
    Frame:        MSO5074
    Label:        CH1
    Serial:       MS5A210900697

    Buffer 1:
        Data Type:      12
        Sample depth:   0 bits
        Buffer length:  256 KB

Segmentation fault
```
The output above suggests some of the data is being misinterpreted - incorrect number of buffers, duration, sample rate, sample depth, etc. There are 1000 8-bit samples in each channel as expected.

Analysis of the minidump shows this is an access violation in [BINImportFilter17OnFileNameChangedEv](https://github.com/ngscopeclient/scopehal/blob/e3a50cc671f52c25da8cc94fb7e4df6efd9f7186/scopeprotocols/BINImportFilter.cpp#L59). I cannot share the .dmp file due to an organisational policy.

WinDbg output:
```

************* Preparing the environment for Debugger Extensions Gallery repositories **************
   ExtensionRepository : Implicit
   UseExperimentalFeatureForNugetShare : true
   AllowNugetExeUpdate : true
   NonInteractiveNuget : true
   AllowNugetMSCredentialProviderInstall : true
   AllowParallelInitializationOfLocalRepositories : true
   EnableRedirectToChakraJsProvider : false

   -- Configuring repositories
      ----> Repository : LocalInstalled, Enabled: true
      ----> Repository : UserExtensions, Enabled: true

>>>>>>>>>>>>> Preparing the environment for Debugger Extensions Gallery repositories completed, duration 0.000 seconds

************* Waiting for Debugger Extensions Gallery to Initialize **************

>>>>>>>>>>>>> Waiting for Debugger Extensions Gallery to Initialize completed, duration 0.032 seconds
   ----> Repository : UserExtensions, Enabled: true, Packages count: 0
   ----> Repository : LocalInstalled, Enabled: true, Packages count: 44

Microsoft (R) Windows Debugger Version 10.0.27829.1001 AMD64
Copyright (c) Microsoft Corporation. All rights reserved.

Loading Dump File [C:\Users\<user>\AppData\Local\CrashDumps\ngscopeclient.exe.18312.dmp]
User Mini Dump File: Only registers, stack and portions of memory are available

************* Path validation summary **************
Response                         Time (ms)     Location
Deferred                                       srv*
Symbol search path is: srv*
Executable search path is: 
Windows 10 Version 26100 MP (20 procs) Free x64
Product: WinNt, suite: SingleUserTS
Edition build lab: 26100.1.amd64fre.ge_release.240331-1435
[... 162 lines cut ...]
000007a0: 807f 7f7f 807f 7f80 807f 807f 7f80 7f7f  ................
000007b0: 7f7f 7f7f 7f7f 7e80 7f7f 807f 7f7f 7f7f  ......~.........
000007c0: 7f80 7f80 8080 807f 7f7f 8080 7f80 807f  ................
000007d0: 7f7f 7f7f 7f7f 7f7f 8080 8080 8080 7f80  ................
000007e0: 7f7e 807f 7f7f 807f 807f 8080 807f 7f7f  .~..............
000007f0: 8080 807f 7f7f 8080 807f 8080 7f80 7f80  ................
00000800: 7f80 807f 7f7f 8080 8080 807f 7e7f 807f  ............~...
00000810: 7f7f 807f 7f80 7f80 807f 7f7f 807f 7f81  ................
00000820: 7f80 807f 8080 7f80 807f 807f 807f 7f80  ................
00000830: 807f 807f 7f80 7f7e 7f7f 7f7f 8080 807f  .......~........
00000840: 807f 7f7f 7f7f 807f 8080 8080 8080 807f  ................
00000850: 7f7f 807f 8080 7f80 807f 8080 7f80 7f80  ................
00000860: 8080 807e 7f80 7f7f 8080 7f80 8080 7f7f  ...~............
00000870: 7f80 8080 807f 807f 8080 7f7f 7f80 7f80  ................
00000880: 7e7f 807f 8080 8080 7f80 807f 7f80 807f  ~...............
00000890: 807f 7f80 7f7f 7f80 8080 8080 8080 8080  ................
000008a0: 7f7f 8080 8080 7f80 8080 8080 7f7f 7e7f  ..............~.
000008b0: 7f80 8080 7f7f 7f80 7f7f 7f7f 7f7f 807f  ................
000008c0: 7f7f 807f 8080 7f7f 807e 7f80 8080 807f  .........~......
000008d0: 8080 7f7f 7f7f 807f 7f7f 7f7f 8080 7f7f  ................
000008e0: 7f80 807f 7f7f 7f7f 7f80 8080 7f7f 7f80  ................
000008f0: 8080 807f 7e80 807f 8080 8080 7f80 807f  ....~...........
00000900: 807f 8080 7f7f 8080 7f7f 807f 807e 807f  .............~..
00000910: 7f7f 8080 7f7f 807f 7f7f 7f80 8080 807f  ................
```

azonenberg, 2025-06-19:
Let me have a look at this.

I can definitely fix the crash, but I'm not sure if we currently support >1 channel of data in a single file. (This would likely be straightforward to add though)

I-AM-EXFOLIATING, 2025-06-19:
I tested again with only a single channel in the file and still observed the crash.

After updating to [scopehal e2b9967](https://github.com/ngscopeclient/scopehal/commit/e2b9967bc9674b5c2c3d2f5e02fab5be5411340f) the crash no-longer occurs, but no waveform data is imported - probably because `dh.depth` is 0.

I-AM-EXFOLIATING, 2025-06-19:
Looking into the data structure it appears the 0x90 WaveHeader size (versus conventional 0x8C) is consistent with the data - there is an extra 4 bytes at the end of the WaveHeader.

[image]

In general the Rigol .bin files seem to be pretty poorly constructed and I'm guessing different models / firmware versions are going to have their own quirks which makes it difficult to support every edge case.

I-AM-EXFOLIATING, 2025-07-15:
After some investigation I think I've identified most of the common issues with .bin exports from the MSO5074.

- @0x04: in the file header, length is always reported as 0x00001048 (4168). The header itself is 168 bytes, i.e. it claims the exported data is always 1000 samples of 4 bytes. This value should be the length of the entire file (header-inclusive).
- @0x08: in the file header, nwav is always 1, regardless of how many channels are exported
- @0x0C: the file header size is correctly reported as 0x00000090 however this is a non-standard header length. Other scopes report a header length 0f 0x0000008C. The extra 4 bytes occur at offset 0x98 and are always 0x00000000. Most software doesn't handle this appropriately.
- @0x18: in the waveheader, num samples is always 1000, regardless of the number of samples in the waveform data.
- @0x20: in the waveheader, duration is always 0x3089705F (1e-9), regardless of the sample duration.
- @0x30: in the waveheader, sample interval is always 0x3D719799 (1e-12), regardless of the sample rate.
- @0xA2: in the data header, sample depth is always 0x0004 (4 bytes), but the data is sampled as uint8.
- @0xA4: in the data header, length is always 0x00000FA0 (4000 bytes), regardless of the payload.
- @0xA8: in the data payload, data is sampled as uint8's, however most software doesn't handle this properly (they are expecting single precision floats)
- Exports of multiple channels simply concatenates multiple .bin files (instead of starting from the waveheader for subsequent channels).

Fixing some of the above issues in a hex editor allows ngscopeclient to import data, however the big dealbreaker is that ngscopeclient ignores the sample depth and assumes the data are single-precision floats, when they are actually raw 8-bit samples (despite the data header claiming 4 bytes per sample).

I'm not sure where you would go from here because the metadata is contradictory and one can't expect the import to work flawlessly if all of this information is untrustworthy. I suggest creating a library of example files for each supported filetype (with different scopes having a sub-type) with manually-curated metadata. This library would serve three purposes - first to develop the importers, second as testcases to prevent regression, and third as a reference for our own files in case we encounter import challenges, since there is no universal published standard for these file formats. If you're interested in this I would be happy to contribute examples from the MSO5074 using the built-in wavegen and probe calibration signals.

azonenberg, 2025-07-15:
A good start would probably be detecting the bogus length at 0x0c which is likely a signature for this buggy file generator, then refusing to load it.

At some point in the future we can think about attempting to reconstruct the damaged files.

azonenberg, 2026-05-25:
Added a warning in the filter graph view when loading these buggy files so users know what's wrong, bumping a more proper fix probably to 0.3 or something