ngscopeclient/scopehal #996: LeCroy WaveSurfer 3054z - Crash on acquisition

rfdonnelly, 2025-08-04:
When trying to acquire waveforms on the WaveSurfer 3054z, ngscopeclient crashes with:

```
terminate called after throwing an instance of 'std: : invalid_argument'
     what():  stoi
Aborted (core dumped)
```

And the scope reports several errors.  I suspect ngscopeclient needs some tweaks to add support for this scope but I'm unsure how to proceed.  Any pointers would be appreciated.

First scope error message:

```
 597.10.2025  20:20:17ErrorAS Interpretererror: command <VBS? app.Acquisition.Horizontal.SampleRate = 2000000> at token <>: parameter cannot be interpreted by this commandcommand <VBS? app.Acquisition.Horizontal.SampleRate = 2000000> at token <>: parameter cannot be interpreted by this command
```

<details>
<summary>Full scope log</summary>

```
07.10.2025  20:19:14InformationalRemote NetConnection accepted from: 169.254.201.128
 17.10.2025  20:19:14InformationalAS Interpreterreceived: *IDN?	
 27.10.2025  20:19:14InformationalAS Interpretersent: LECROY,WS3054Z,LCRY3710N45420,11.1.0
 37.10.2025  20:19:14InformationalAS Interpreterreceived: CHDR OFF
 47.10.2025  20:19:14InformationalAS Interpreterreceived: *IDN?	
 57.10.2025  20:19:14InformationalAS Interpretersent: LECROY,WS3054Z,LCRY3710N45420,11.1.0
 67.10.2025  20:19:14InformationalAS Interpreterreceived: COMM_FORMAT DEF9,BYTE,BIN
 77.10.2025  20:19:14InformationalAS Interpreterreceived: INR?	
 87.10.2025  20:19:14InformationalAS Interpretersent: 0
 97.10.2025  20:19:14InformationalAS Interpreterreceived: *OPT?	
 107.10.2025  20:19:14InformationalAS Interpretersent: I2C,SPI,UART,AFG
 117.10.2025  20:19:14InformationalAS Interpreterreceived: COMM_FORMAT DEF9,WORD,BIN
 127.10.2025  20:19:14InformationalAS Interpreterreceived: C1:TRACE?	
 137.10.2025  20:19:14InformationalAS Interpretersent: ON
 147.10.2025  20:19:14InformationalAS Interpreterreceived: C2:TRACE?	
 157.10.2025  20:19:14InformationalAS Interpretersent: OFF
 167.10.2025  20:19:15InformationalAS Interpreterreceived: C3:TRACE?	
 177.10.2025  20:19:15InformationalAS Interpretersent: OFF
 187.10.2025  20:19:15InformationalAS Interpreterreceived: C4:TRACE?	
 197.10.2025  20:19:15InformationalAS Interpretersent: OFF
 207.10.2025  20:19:15InformationalAS Interpreterreceived: C1:TRACE ON
 217.10.2025  20:19:15InformationalAS Interpreterreceived: VBS? 'return = app.Acquisition.Trigger.Type'	VBS? 'return = app.Acquisition.Trigger.Type'
 227.10.2025  20:19:15InformationalAS Interpretersent: Edge
 237.10.2025  20:19:15InformationalAS Interpreterreceived: VBS? 'return = app.Acquisition.Trigger.Edge.Level'	VBS? 'return = app.Acquisition.Trigger.Edge.Level'
 247.10.2025  20:19:15InformationalAS Interpretersent: -1.04
 257.10.2025  20:19:15InformationalAS Interpreterreceived: VBS? 'return = app.Acquisition.Trigger.Edge.Slope'	VBS? 'return = app.Acquisition.Trigger.Edge.Slope'
 267.10.2025  20:19:15InformationalAS Interpretersent: Positive
 277.10.2025  20:19:15InformationalAS Interpreterreceived: VBS? 'return = app.Acquisition.Trigger.Source'	VBS? 'return = app.Acquisition.Trigger.Source'
 287.10.2025  20:19:15InformationalAS Interpretersent: C1
 297.10.2025  20:19:15InformationalAS Interpreterreceived: VBS? 'return = app.Acquisition.C1.Invert'	VBS? 'return = app.Acquisition.C1.Invert'
 307.10.2025  20:19:15InformationalAS Interpretersent: 0
 317.10.2025  20:19:15InformationalAS Interpreterreceived: TRDL?	
 327.10.2025  20:19:15InformationalAS Interpretersent: 0E-3
 337.10.2025  20:19:16InformationalAS Interpreterreceived: VBS? 'return = app.Acquisition.Horizontal.SamplingRate'	VBS? 'return = app.Acquisition.Horizontal.SamplingRate'
 347.10.2025  20:19:16InformationalAS Interpretersent: 1000000
 357.10.2025  20:19:16InformationalAS Interpreterreceived: VBS? 'return = app.Acquisition.Horizontal.AcquisitionDuration'	VBS? 'return = app.Acquisition.Horizontal.AcquisitionDuration'
 367.10.2025  20:19:16InformationalAS Interpretersent: 0.1
 377.10.2025  20:19:16InformationalAS Interpreterreceived: C1:OFFSET?	
 387.10.2025  20:19:16InformationalAS Interpretersent: -2.41
 397.10.2025  20:19:16InformationalAS Interpreterreceived: C1:VOLT_DIV?	
 407.10.2025  20:19:16InformationalAS Interpretersent: 1
 417.10.2025  20:19:16InformationalAS Interpreterreceived: INR?	
 427.10.2025  20:19:16InformationalAS Interpretersent: 0
 437.10.2025  20:19:16InformationalAS Interpreterreceived: VBS? 'return = app.WaveSource.enable'	
 447.10.2025  20:19:16InformationalAS Interpretersent: 0
 457.10.2025  20:19:16InformationalAS Interpreterreceived: VBS? 'return = app.WaveSource.Amplitude'	VBS? 'return = app.WaveSource.Amplitude'
 467.10.2025  20:19:16InformationalAS Interpretersent: 0.1
 477.10.2025  20:19:16InformationalAS Interpreterreceived: VBS? 'return = app.WaveSource.Offset'	
 487.10.2025  20:19:16InformationalAS Interpretersent: 0
 497.10.2025  20:19:16InformationalAS Interpreterreceived: VBS? 'return = app.WaveSource.Frequency'	VBS? 'return = app.WaveSource.Frequency'
[... 92 lines cut ...]
 1427.10.2025  20:22:25InformationalAS Interpreterreceived: VBS? 'return = app.Acquisition.Trigger.Edge.Level'	VBS? 'return = app.Acquisition.Trigger.Edge.Level'
 1437.10.2025  20:22:25InformationalAS Interpretersent: -1.04
 1447.10.2025  20:22:25InformationalAS Interpreterreceived: VBS? 'return = app.Acquisition.Trigger.Edge.Slope'	VBS? 'return = app.Acquisition.Trigger.Edge.Slope'
 1457.10.2025  20:22:25InformationalAS Interpretersent: Positive
 1467.10.2025  20:22:25InformationalAS Interpreterreceived: VBS? 'return = app.Acquisition.Trigger.Source'	VBS? 'return = app.Acquisition.Trigger.Source'
 1477.10.2025  20:22:25InformationalAS Interpretersent: C1
 1487.10.2025  20:22:25InformationalAS Interpreterreceived: VBS? 'return = app.Acquisition.C1.Invert'	VBS? 'return = app.Acquisition.C1.Invert'
 1497.10.2025  20:22:25InformationalAS Interpretersent: 0
 1507.10.2025  20:22:25InformationalAS Interpreterreceived: TRDL?	
 1517.10.2025  20:22:25InformationalAS Interpretersent: 0E-3
 1527.10.2025  20:22:25InformationalAS Interpreterreceived: VBS? 'return = app.Acquisition.Horizontal.SamplingRate'	VBS? 'return = app.Acquisition.Horizontal.SamplingRate'
 1537.10.2025  20:22:25InformationalAS Interpretersent: 1000000
 1547.10.2025  20:22:25InformationalAS Interpreterreceived: VBS? 'return = app.Acquisition.Horizontal.AcquisitionDuration'	VBS? 'return = app.Acquisition.Horizontal.AcquisitionDuration'
 1557.10.2025  20:22:25InformationalAS Interpretersent: 0.1
 1567.10.2025  20:22:25InformationalAS Interpreterreceived: C1:OFFSET?	
 1577.10.2025  20:22:25InformationalAS Interpretersent: -2.41
 1587.10.2025  20:22:25InformationalAS Interpreterreceived: C1:VOLT_DIV?	
 1597.10.2025  20:22:25InformationalAS Interpretersent: 1
 1607.10.2025  20:22:25InformationalAS Interpreterreceived: VBS? 'return = app.Acquisition.Horizontal.SampleMode'	VBS? 'return = app.Acquisition.Horizontal.SampleMode'
 1617.10.2025  20:22:25InformationalAS Interpretersent: RealTime
 1627.10.2025  20:22:25InformationalAS Interpreterreceived: VBS? 'return = app.Acquisition.Horizontal.SampleMode'	VBS? 'return = app.Acquisition.Horizontal.SampleMode'
 1637.10.2025  20:22:25InformationalAS Interpretersent: RealTime
```

</details>

azonenberg, 2025-08-04:
Do you have a Windows PC or VM that you can install https://www.teledynelecroy.com/support/softwaredownload/wavestudio.aspx in?

It should, if memory serves me right, install the XStream Browser which will let you connect to the scope and view the COM object hierarchy, then we can figure out what's different about it vs what the driver expects.

(The higher end LeCroy scopes running full Windows include the browser on them, but the WS3K and I believe also 3000Z and 4000HD are Windows CE based and do not support running the browser on the scope itself)

azonenberg, 2025-08-04:
As a near term fix I can add some sanity checking or try/catch around the stoi call to make the driver not crash, but some things will definitely not be happy until we fix these errors.

We have a WaveSurfer 3034 (non-Z) at work and I'll check next time I'm there if it has any issues, but it's probably going to be a while. I mostly work out of my home lab which has a WaveRunner 8404M-MS and a SDA816Zi-A, both of which run this driver with no issues. You may be the first to try it on a 3000z.

rfdonnelly, 2025-08-06:
I was unable to use XStream Browser.  It doesn't seem to be compatible with this class of scope.  I had to use the Automation Browser within WaveStudio instead.  It didn't allow me to dump the model.  Instead, I captured two screenshots of the relevant items and a packet capture ([scopehal-issue-996.pcapng.tar.gz](https://github.com/user-attachments/files/21609208/scopehal-issue-996.pcapng.tar.gz)) of the exchange.

You can use wireshark to view the reassembled tcp stream to get the model xml (example screenshot below) by selecting a vicp packet, then select the analyze>follow>tcp stream menu item.  I can also extract the XML into a nicely formatted XML file.  Let me know if you would like that.

Acquisition.C1
[image]
NOTE: Missing `Acquisition.C1.ActiveInput`, `Acquisition.C1.10.*`, and `Acquisition.C1.AverageSweeps`.  However, `AverageSweeps` is present under `Acquisition.Horizontal`.

Acquisition.Horizontal
[image]

Wireshark TCP Stream example:
[image]

> Does your scope have a hardware/firmware averaging capability under channel settings?

I'm not 100% sure.  There is some averaging capability under math but there doesn't seem to be under channel settings.  And there is `AverageSweeps` under `Acquisition.Horizontal` but not under `Acquisition.C1`.