ngscopeclient/scopehal #1002: "file read error" in log viewer when opening a csv file on windows10 64bit LTSC with ngscopeclient v0.1 github release build

wchgod001-gh, 2025-09-19:
i can successfully opening .bin file (saved from a keysight scope), but failed to open a .csv file ( saved from a dreamsouce scope, manually modified, as it's data format is not directly supported ).
i have compared the code for file opening in` lib\scopeprotocols\CSVImportFilter.cpp` and `lib\scopeprotocols\BINImportFilter.cpp`, it's the same.
and tried to extract this piece of code into a cpp file, compiled and executed successfully with no error.

```
//Read the entire file into a buffer
	FILE* fp = fopen(fname.c_str(), "r");
	if(!fp)
	{
		LogError("Couldn't open CSV file \"%s\"\n", fname.c_str());
		return;
	}
	fseek(fp, 0, SEEK_END);
	size_t flen = ftell(fp);
	fseek(fp, 0, SEEK_SET);
	char* buf = new char[flen+1];
	if(flen != fread(buf, 1, flen, fp))
	{
		LogError("file read error\n");
		return;
	}
	buf[flen] = '\0';	//guarantee null termination at end of file
	fclose(fp);
```

i guess the content of this csv file is not relevant, what could possibly be wrong?  

[out-100line.csv](https://github.com/user-attachments/files/22423581/out-100line.csv)

azonenberg, 2025-09-19:
My first guess is that it has something to do with Windows vs UNIX line endings and reading the entire file into memory.

Maybe ftell() is reporting the length in bytes, but we're trying to read in characters and line endings are confusing it. Try changing the "r" in fopen to "rb" to read in binary mode and see if it works then.

azonenberg, 2025-09-19:
(but if we read in binary mode we will possibly have to patch some of the parsing code to ignore both \r and \n)

commit 4ff5f916 (2025-09-20): CSVImportFilter: open files in binary mode and explicitly handle \r for Windows line endings. Fixes #1002.
closed 2025-09-20 by azonenberg